cortex-lab/phy #899: table entries and labels in giant font

jenniferColonell, 2019-09-25:
On several different laptops (tested at the KS2 workshop and another course) we've found that the cluster table is displayed in huge font, behind the other views -- see attached screenshot. In this case, running on a Windows 10 laptop with onboard graphics only (Intel UHD Graphics 620), OpenGL version = 4.6.
[image]

jenniferColonell, 2019-09-26:
Just some more details:
running with --debug shows no error messages
tried changing the font family in style.css for the table from Helvetica to Arial --> no change
tried updating the driver for the graphics --> no change

We saw this problem on about 25% of the laptops used in the KS2 course last month.  So it would be really useful to learn a fix!

rossant, 2019-09-26:
Where is the cluster table in this screenshot?
It would be quite helpful if you could give the configuration of laptops where there is a problem and those where there isn't. At least OS, graphics card, screen resolution (high dpi or not)

jenniferColonell, 2019-09-26:
The cluster table is sort of overlaid over the other views -- the super big "ID" and "Cluster" labels are visible in the upper right.

I'll try to gather some data about good and bad laptops.

rossant, 2019-09-26:
Ah, I see. The display seems seriously broken! I'll see if I can reproduce this on a Windows machine with Intel GPU. Curious to know if this also occurs with an NVIDIA GPU.

jenniferColonell, 2019-10-03:
Another clue:
looks like laptops with nvidia graphics are OK
laptops with older Intel HD graphics are OK
laptops with new Intel UHD graphics (I've only tested with Intel UHD Graphics 620) have a problem.

frikyng, 2019-12-02:
Hi, 
I had the exact same problem with my PC, which also has Intel UHD Graphics 620 onboard. However, there is another, discrete, graphics card built in (AMD Radeon RX 550X). I could solve the problem by running Python on this other card. For this I used the software that AMD provides (MD Radeon™ Software Adrenalin 2019 Edition).
If you have dedicated Nvidia graphics, they have the NVIDIA control panel, which in theory should be able to do the same thing. Hope this works for you!
In the image you can see that GPU 1 instead of GPU 0 is ran by Python, indicating usage of the dedicated card. 

[image]

rossant, 2019-12-04:
thanks @frikyng

closed 2019-12-04 by rossant