cortex-lab/phy #901: Channels not displayed in linear electrode (1x32) in Phy2 only

elyxa, 2019-09-30:
Hi everybody,

I am trying to visualize data recorded with a 1x32 shaft and sorted with Kilosort1.
If I open the data in Phy1 I can visualize all the channels correctly.
With Phy2, whatever spike sorted 1x32 recordings I open, I cannot visualize some of the channels. The view always looks as follows (figure on top), with some channels always not displayed and others just displaying a line, while what we expect to see is the visualization we have in Phy1 (figure on the bottom). The same cluster is visualized in both views.

[image]

[image]

We have the same issue in both computers of our lab having Phy2 installed.
With 4x8 electrodes we don't have any issue, so the problem seems to be related to the 1x32 channel mapping.

Do you know what could be the reason and how I could solve this issue?
Otherwise, is there any way to download the old version of Phy?

Thank you in advance for your help!

Best wishes,
Elisabetta

rossant, 2019-10-07:
Hello, could you send me your dataset? This particular bug should have been fixed recently but I'd like to double-check with your data. Thanks!

rossant, 2019-12-24:
should be fixed

closed 2019-12-24 by rossant