cortex-lab/phy #905: System cannot find the file specified

riossa, 2019-10-09:
I'm fairly new to python, when downloading, I typed the last two lines specified in the download instructions
(pip install git+https://github.com/cortex-lab/phy.git@dev
pip install git+https://github.com/cortex-lab/phylib.git)

anaconda returned 2 errors
1) file specified cannot be found
2) cannot find command 'git' - do you have 'git' installed and in your path

rossant, 2019-10-10:
You probably need to install git, see https://git-scm.com/download/win

riossa, 2019-10-10:
> Or otherwise, you can use these command lines without installing git
>
> pip install --upgrade
> https://github.com/cortex-lab/phylib/archive/master.zip
> <http://github.com/cortex-lab/phylib/archive/master.zip>
> pip install --upgrade https://github.com/cortex-lab/phy/archive/dev.zip
> <http://github.com/cortex-lab/phy/archive/master.zip>

Hi Pierre,

Thank you, that solved that issue. Unfortunately now when I get to the last step “phy template- gui params.py” I get the error "invalid value for "PARAMS_PATH": Path "params.py" does not exist.”

I made sure to add GIT to my path like I saw on the forums and made sure everything else downloaded correctly which it did….

If you could help with this issue as well that would be much appreciated,

Thank you,
Samantha Rios

riossa, 2019-10-10:
Hi Cyrille,

Thank you, that solved that issue. Unfortunately now when I get to the last step “phy template- gui params.py” I get the error "invalid value for "PARAMS_PATH": Path "params.py" does not exist.”

I made sure to add GIT to my path like I saw on the forums and made sure everything else downloaded correctly which it did….

If you could help with this issue as well that would be much appreciated,

Thank you,
Samantha Rios

riossa, 2019-10-14:
> When you run the command, are you in a directory that contains your data, including the params.py file?

So I'm following the instructions on https://phy.readthedocs.io/en/latest/installation/ . The first time i see params is on the step thats giving me issues (phy template-gui params.py). Which step is it that adds params/ or how/where does it add it to the computer because I'm not finding it.

Thanks

rossant, 2019-10-14:
You need to run this command in a directory that contains a dataset in order to open this dataset in the GUI. Typically, you obtain this dataset with an automated spike sorting program such as KiloSort or Spyking Circus, which takes a multichannel electrophysiological recording as input, and returns automatically-detected spikes. To test the GUI, you can also download a test dataset in this repository: https://github.com/kwikteam/phy-data

riossa, 2019-11-14:
Hi Cyrille,

I recently got a new computer and had to reinstall everything for the GUI. For some reason though, I can get all the way to the GUI appearing with the relevant information but the waveforms do not come up… I’ve attached a photo for clarification (GUI). I can open it no problem on another computer so I’ve tried uninstalling everything and trying again with no luck. I’ve gone into the phy text file to see what the issue is but the error codes I’m getting aren’t much helpful (I’ve attached that file as well) . I’ve tried both phy and phy2, GIT is installed, and I’m using anaconda PowerShell.

If you have any idea what the issue is, your help would be greatly appreciated,
Thanks,
Samantha

rossant, 2019-11-15:
Hi, I'm not seeing the screenshot nor the log file. Could you send me directly to my email address? Thanks

riossa, 2019-11-15:
Yes of course, if you could just provide me with your email.

closed 2019-12-24 by rossant