cortex-lab/phy #906: phy2 waveform view TypeError

tonylien, 2019-10-13:
Hello,

I installed phy2 and am having problems with the waveform view. When I click a cluster, the waveforms don't appear and I get 

> TypeError: pad() missing 1 required positional argument: 'mode'

```
Traceback (most recent call last):
  File "c:\miniconda3\envs\phy2\lib\site-packages\phy\gui\qt.py", line 453, in run
    result = self.fn(*self.args, **self.kwargs)
  File "c:\miniconda3\envs\phy2\lib\site-packages\phy\cluster\views\base.py", line 158, in _worker
    self.on_select(cluster_ids=cluster_ids, **kwargs)
  File "c:\miniconda3\envs\phy2\lib\site-packages\phy\cluster\views\base.py", line 130, in on_select
    self.plot(**kwargs)
  File "c:\miniconda3\envs\phy2\lib\site-packages\phy\cluster\views\waveform.py", line 292, in plot
    bunchs = self.get_clusters_data()
  File "c:\miniconda3\envs\phy2\lib\site-packages\phy\cluster\views\waveform.py", line 172, in get_clusters_data
    self.waveforms[self.waveforms_type](cluster_id) for cluster_id in self.cluster_ids]
  File "c:\miniconda3\envs\phy2\lib\site-packages\phy\cluster\views\waveform.py", line 172, in <listcomp>
    self.waveforms[self.waveforms_type](cluster_id) for cluster_id in self.cluster_ids]
  File "c:\miniconda3\envs\phy2\lib\site-packages\phy\apps\base.py", line 210, in _get_waveforms
    current_filter=self.raw_data_filter.current_filter)
  File "c:\miniconda3\envs\phy2\lib\site-packages\joblib\memory.py", line 568, in __call__
    return self._cached_call(args, kwargs)[0]
  File "c:\miniconda3\envs\phy2\lib\site-packages\joblib\memory.py", line 534, in _cached_call
    out, metadata = self.call(*args, **kwargs)
  File "c:\miniconda3\envs\phy2\lib\site-packages\joblib\memory.py", line 734, in call
    output = self.func(*args, **kwargs)
  File "c:\miniconda3\envs\phy2\lib\site-packages\phy\apps\base.py", line 194, in _get_waveforms_with_n_spikes
    data = self.model.get_waveforms(spike_ids, channel_ids)
  File "c:\miniconda3\envs\phy2\lib\site-packages\phylib\io\model.py", line 823, in get_waveforms
    w = np.pad(w, ((bef, aft), (0, 0)))
TypeError: pad() missing 1 required positional argument: 'mode'
```

All the other views seem to be correct. Subsequent selection of different clusters causes more errors and eventually it crashes. Also, if I press "w" after the initial error, the templates do appear in the waveform view. I get the error on my own data sorted with kilosort1 as well as with the test data set from here: https://github.com/kwikteam/phy-data/tree/master/template. I had no problems opening the same data in phy1.

I installed phy2 according to:  https://phy.readthedocs.io/en/latest/installation/ and also ran the upgrade commands: 
```
pip install --upgrade git+https://github.com/cortex-lab/phy.git@dev
pip install --upgrade git+https://github.com/cortex-lab/phylib.git
```
The full dump from the command window is below. Any help is greatly appreciated!

```
(phy2) PS F:\kilosortdata\phytestdata> phy template-gui --debug --clear-state params.py
←[90m01:37:53.680 [D] __init__:68          Start capturing exceptions.←[0m
←[90m01:37:54.085 [D] model:526            Loading spike clusters.←[0m
←[90m01:37:54.092 [D] model:575            Loading templates.←[0m
←[90m01:37:54.095 [D] model:598            Templates are sparse.←[0m
←[90m01:37:54.095 [D] model:607            Loading the whitening matrix.←[0m
←[90m01:37:54.096 [D] model:614            Loading the inverse of the whitening matrix.←[0m
←[90m01:37:54.101 [D] model:159            Loading traces at `F:\kilosortdata\phytestdata\sim_binary.dat`.←[0m
←[90m01:37:54.103 [D] model:643            Loading features.←[0m
←[90m01:37:54.105 [D] model:658            Features are sparse.←[0m
←[90m01:37:54.106 [D] model:680            Loading template features.←[0m
←[90m01:37:54.109 [D] model:690            Template features are sparse.←[0m
←[90m01:37:54.112 [D] model:413            Load `F:\kilosortdata\phytestdata\cluster_group.tsv`.←[0m
←[90m01:37:54.113 [D] _misc:319            Read F:\kilosortdata\phytestdata\cluster_group.tsv.←[0m
←[90m01:37:54.113 [D] context:80           Create cache directory `F:\kilosortdata\phytestdata\.phy`.←[0m
←[90m01:37:54.382 [D] context:100          Initialize joblib cache dir at `F:\kilosortdata\phytestdata\.phy`.←[0m
←[90m01:37:54.384 [D] context:101          Reducing the size of the cache if needed.←[0m
←[33m01:37:54.386 [W] base:876             Deleting C:\Users\tony.lien\.phy\TemplateGUI\state.json.←[0m
←[90m01:37:54.389 [D] config:31            Load config file `C:\Users\tony.lien\.phy\phy_config.py`.←[0m
←[90m01:37:54.391 [D] plugin:145           Loading 0 plugins.←[0m
←[90m01:37:54.401 [D] context:209          The file `F:\kilosortdata\phytestdata\.phy\new_cluster_id.pkl` doesn't exist.←[0m
←[90m01:37:54.405 [D] context:209          The file `F:\kilosortdata\phytestdata\.phy\spikes_per_cluster.pkl` doesn't exist.←[0m
←[90m01:37:54.407 [D] clustering:237       Recompute spikes_per_cluster manually: this might take a while.←[0m
←[90m01:37:54.410 [D] context:185          Save data to `F:\kilosortdata\phytestdata\.phy\spikes_per_cluster.pkl`.←[0m
←[90m01:37:54.420 [D] gui:337              Creating GUI.←[0m
[... 33 lines cut ...]
  File "c:\miniconda3\envs\phy2\lib\site-packages\phy\cluster\views\waveform.py", line 292, in plot
    bunchs = self.get_clusters_data()
  File "c:\miniconda3\envs\phy2\lib\site-packages\phy\cluster\views\waveform.py", line 172, in get_clusters_data
    self.waveforms[self.waveforms_type](cluster_id) for cluster_id in self.cluster_ids]
  File "c:\miniconda3\envs\phy2\lib\site-packages\phy\cluster\views\waveform.py", line 172, in <listcomp>
    self.waveforms[self.waveforms_type](cluster_id) for cluster_id in self.cluster_ids]
  File "c:\miniconda3\envs\phy2\lib\site-packages\phy\apps\base.py", line 210, in _get_waveforms
    current_filter=self.raw_data_filter.current_filter)
  File "c:\miniconda3\envs\phy2\lib\site-packages\joblib\memory.py", line 568, in __call__
    return self._cached_call(args, kwargs)[0]
  File "c:\miniconda3\envs\phy2\lib\site-packages\joblib\memory.py", line 534, in _cached_call
    out, metadata = self.call(*args, **kwargs)
  File "c:\miniconda3\envs\phy2\lib\site-packages\joblib\memory.py", line 734, in call
    output = self.func(*args, **kwargs)
  File "c:\miniconda3\envs\phy2\lib\site-packages\phy\apps\base.py", line 194, in _get_waveforms_with_n_spikes
    data = self.model.get_waveforms(spike_ids, channel_ids)
  File "c:\miniconda3\envs\phy2\lib\site-packages\phylib\io\model.py", line 823, in get_waveforms
    w = np.pad(w, ((bef, aft), (0, 0)))
TypeError: pad() missing 1 required positional argument: 'mode'
←[33m01:38:07.132 [W] variable:389         Data is empty for a_box_index←[0m
←[33m01:38:07.133 [W] variable:389         Data is empty for a_color←[0m
←[33m01:38:07.133 [W] variable:389         Data is empty for a_position←[0m
←[33m01:38:07.143 [W] variable:389         Data is empty for a_size←[0m
←[90m01:38:07.144 [D] base:655             Exception in paintGL: 'numpy.ndarray' object has no attribute 'set_interpolation'←[0m
```

rossant, 2019-10-14:
Thanks for the bug report. I pushed a tentative fix on phylib. Could you update phylib with `pip install --upgrade git+https://github.com/cortex-lab/phylib.git` and try again?

tonylien, 2019-10-14:
Looks like that fixed it. The WaveformView is working now! Thanks!

closed 2019-10-14 by rossant