cortex-lab/phy #907: Run phy2 in windows subsystem for linux (WSL)

sina253, 2019-10-18:
Hi

I've tried to run the phy2 in the WSL and got the following errors:

[image]

rossant, 2019-10-21:
Do you have a `spike_templates.npy` file in your data directory?

sina253, 2019-10-21:
Hi
Thank you,
No there was no spike_templates.npy file in data directory.  Now I copied all the files in the data directory. Also installed Anaconda3 and phy2 on the Windows10 directly not on WSL.
Now I get new below error (there is template_features.npy in the data direcory):

[image]

closed 2019-12-24 by rossant