cortex-lab/phy #908: waveform view scaling difference between raw and mean or template

tonylien, 2019-10-25:
I'm experiencing similar scaling discrepancies between the raw and mean/templates in the waveform view as #850 . I have upgraded to the latest version of phy2 and the problem persists. It seems like units with smaller raw amplitudes have much larger amplitude mean/templates while larger raw units have less discrepancy. 

Small amplitude unit raw, mean, and template:
[image]
[image]
[image]

Large amplitude unit raw, mean, and template::
[image]
[image]
[image]

nsteinme, 2019-11-15:
Agreed, we also observed this and it seems like a fast and urgent fix - it's very confusing to a lot of people! Thanks @tonylien

rossant, 2019-11-16:
A partial fix has been pushed, such that the same scaling is used on all types of waveforms, for all clusters. But there's a new problem, which is that the discrepancy between the raw waveforms and the template waveforms is even more apparent. It appears that multiplying the templates by the inverse of the whitening matrix (even though it normally includes the scaleproc parameter of KS2) does not match the raw data. I don't know why. In the meantime, I multiply the templates by a magic value of 20 and it seems to approximately match on the datasets I've tried. We should look more into it to understand where this discrepancy comes from.

nsteinme, 2019-11-18:
Just to clarify, are these old datasets you have sitting around, or ones
you have run newly through latest ks2?

rossant, 2019-11-18:
Both

rossant, 2019-11-18:
cc @marius10p

rossant, 2019-12-24:
The discrepancy likely comes from KS2, until this is fixed upstream a work-around in phy is to add the line `template_scaling = 20.0` in `params.py` (or with any other scaling factor that empirically works on a given dataset)

closed 2019-12-24 by rossant