cortex-lab/phy #910: alternative amplitude views deactivated?

marius10p, 2019-11-07:
I am in the dev version, and I can no longer cycle through amplitudes, which I assume means there is only one type of amplitude, the one directly provided by kilosort. Previously, Phy2 was computing the "raw" amplitudes and those were very useful to complement the kilosort "template" amplitudes.

rossant, 2019-11-07:
This is weird, any error message in the terminal? this is a dataset where you have the raw data available, and both waveform view and trace view? Can you cycle through the different types of waveforms as well?

marius10p, 2019-11-07:
I can cycle through waveforms, and I don't get an error message. Let me
check a different dataset...

marius10p, 2019-11-08:
What happened was that I had two amplitude views open and it was cycling the primary one. However, the primary amplitude view is nowhere to be found! Together with a few other views. That's why I opened new ones. I attach a screenshot (they are not undocked either). Maybe a button in the drop-down menu to close a view would be good? 

[image]

rossant, 2019-11-08:
OK, in the meantime you can look at `~/.phy/TemplateGUI/state.json` and change the view_count accordingly. Or just delete this file altogether to reset the default.

closed 2019-11-16 by rossant