cortex-lab/phy #911: missing phy2 waveform

boubenec, 2019-11-07:
Hi,
I'm doing my clustering with klusta. For some reasons, when I open the .kwik file with phy2 (phy kwik-gui), I miss the waveforms (all channels are flat), although all the other panels (trace, correlogram, features and so forth) work normally for any selected cluster.
Is it something that have been reported previously?
Thanks for the help. Cheers, Yves.

rossant, 2019-11-08:
Could you delete the `~/.phy/` directory in your home directory, as well as the .phy subfolder in your data directory, and try again? It might be a bad waveform scaling issue.

boubenec, 2019-11-12:
Amazing, it's working. The waveforms are displayed! Thanks sooo much. Yves.

closed 2019-12-24 by rossant