cortex-lab/phy #912: Problem installing PHY2 on a Windows10 machine

lalittaSA, 2019-11-08:
Hi, 
I've to install PHY2 on a new computer but once I completed all the processes and wanted to open a file, I got these error messages. What did I do wrong? Can there be version issue?

Thanks for your support,
Lalitta
[image]
[image]
[image]
[image]

rossant, 2019-11-08:
It does look like a conflict between multiple versions. Ensure that all previous versions are uninstalled, and delete ̀~/.phy` in your home directory, as well as the .phy subfolders that may exist in your data directories.

lalittaSA, 2019-11-15:
Thank you for your reply. Now I get another error message regarding python version. Just to confirm, do I need Python 2.7, right? (I installed Anaconda 3 and PHY complained).

Many thanks,
L

lalittaSA, 2019-11-15:
Forgot the screenshot.

[image]

lalittaSA, 2019-11-18:
I removed everything and reinstalled Anaconda and PHY2 from scratch but still got the same issue.
Which Python version is installed in your virtual env, please? I tried python 3.7 and 3.6 they gave the same error message (that I was trying to read with python 3 a joblib pickle generated with python 2, as shown in the above screenshot). However, with python 3.5, I got a different error message regarding the path (see screenshot). Any suggestions? 

Thank you so much
[image]

rossant, 2019-11-20:
I use Python 3.7. It seems that the .phy subfolder was not properly removed. Make sure you enable the display of system and cached folders in your explorer settings? The directory you want to delete is `I:\kiloSorted\MrCassius-190324\D1_PFC_R1\KS2_7\.phy`.

lalittaSA, 2019-11-20:
Ohh!! Now it works. I thought I was supposed to remove .phy just in the folder I installed Anaconda. Thank you so much again for your kind support.

closed 2019-12-24 by rossant