cortex-lab/phy #913: Easier installation via pip

simon-ball, 2019-11-08:
`setup.py` does not specify the list of requirements, and therefore `pip install git+https://github.com/cortex-lab/phy` does not install missing dependencies. Instead, it requires 3 steps (as shown in the readme: clone, cd, install requirements). 

Is this a deliberate choice (e.g. to enourage users to set up a relevant virtual envronment), or an oversight? 

If the latter I would be happy to submit a fix

rossant, 2019-11-08:
A problem I had with specifying the dependencies in the setup is that, sometimes, pip tries to install NumPy and SciPy and other packages that are already installed with conda, or that should really be installed via conda and not pip. Otherwise, having a mix of pip and conda compiled packages leads to conflicts and crashes. Do you know how to avoid this problem?

simon-ball, 2019-11-08:
Hrmm. The issue is ameliorated somewhat as nearly none of the requirements for `phy` are pinned to specific versions (only `mkdocs` is). Pip should only attempt to install e.g., Numpy if there is no version at all present in the current environment: if a specific version, even an ancient one, has been installed via `conda`, the requirement for Numpy should be satisfied. 

I've not yet come across a scenario in which `pip` tries to reinstall requirements which are not pinned (although that's hardly a guarantee that it cannot happen). I do not have a solution to the `pip`-`conda` interface, no. 

If you have specific past reasons to stick with the current approach, then fair enough - I was mostly asking on the offchance that it was an oversight instead of a deliberate choice.

rossant, 2019-11-08:
When I had this problem, it was indeed a case where a specific version was pinned in the requirement file, so that might be fine. I would be happy to consider a pull request, thanks!

simon-ball, 2019-11-12:
Question about the `phylib` requirement: it is given as fetching the latest version from the master branch of https://github.com/cortex-lab/phylib. The version of `phylib` on `pip` appears to have the same version number (although presumably is typically a few commits behind).

That's not a problem for `pip install -r requirements.txt`, since `pip` understands that as `pip install <line1> <line2>` etc, but requires additional steps to be used in `install_requires`:
* Add the github repo to `depedency_links=[]`
* Call `pip install <package> --process-dependency-links` to have `pip` reocgnise the correct source of the package

So, question: is it a critical requirement that `phylib` fetch the most recent version from github, instead of the latest version published to `pip`?

rossant, 2019-11-13:
@oliche is now making proper releases of phylib so I think we can switch to depend on the pip package of phylib instead
soon enough we should also make a proper release of phy..

rossant, 2020-01-22:
fixed in the next release

closed 2020-01-22 by rossant