cortex-lab/phy #915: Assertion error in the number of cluster

julieelie, 2019-11-12:
Hello,
I'm new to Kilosort2 and Phy and randomly getting issues when loading output data from kilosort2 to Phy. Could anyone point me to what could cause that problem?
I'm trying to spike sort long recordings from 4 tetrodes and I'm attaching [here](https://drive.google.com/drive/folders/1LnqNvIuV11ZxoJQo9Yw7iCVB5-QLzgI4?usp=sharing) the npy and tsv files that give me issues.
Thank you for your help!
Julie

Here is the error:

`Traceback (most recent call last):
  File "/Users/elie/anaconda3/envs/phy2/bin/phy", line 8, in <module>
    sys.exit(phycli())
  File "/Users/elie/anaconda3/envs/phy2/lib/python3.7/site-packages/click/core.py", line 764, in __call__
    return self.main(*args, **kwargs)
  File "/Users/elie/anaconda3/envs/phy2/lib/python3.7/site-packages/click/core.py", line 717, in main
    rv = self.invoke(ctx)
  File "/Users/elie/anaconda3/envs/phy2/lib/python3.7/site-packages/click/core.py", line 1137, in invoke
    return _process_result(sub_ctx.command.invoke(sub_ctx))
  File "/Users/elie/anaconda3/envs/phy2/lib/python3.7/site-packages/click/core.py", line 956, in invoke
    return ctx.invoke(self.callback, **ctx.params)
  File "/Users/elie/anaconda3/envs/phy2/lib/python3.7/site-packages/click/core.py", line 555, in invoke
    return callback(*args, **kwargs)
  File "/Users/elie/anaconda3/envs/phy2/lib/python3.7/site-packages/click/decorators.py", line 17, in new_func
    return f(get_current_context(), *args, **kwargs)
  File "/Users/elie/anaconda3/envs/phy2/lib/python3.7/site-packages/phy/apps/__init__.py", line 156, in cli_template_gui
    template_gui(params_path, **kwargs)
  File "/Users/elie/anaconda3/envs/phy2/lib/python3.7/site-packages/phy/apps/template/gui.py", line 195, in template_gui
    controller = TemplateController(**get_template_params(params_path), **kwargs)
  File "/Users/elie/anaconda3/envs/phy2/lib/python3.7/site-packages/phy/apps/base.py", line 438, in __init__
    super(TemplateMixin, self).__init__(*args, **kwargs)
  File "/Users/elie/anaconda3/envs/phy2/lib/python3.7/site-packages/phy/apps/base.py", line 854, in __init__
    self._set_supervisor()
  File "/Users/elie/anaconda3/envs/phy2/lib/python3.7/site-packages/phy/apps/template/gui.py", line 95, in _set_supervisor
    super(TemplateController, self)._set_supervisor()
  File "/Users/elie/anaconda3/envs/phy2/lib/python3.7/site-packages/phy/apps/base.py", line 952, in _set_supervisor
    context=self.context,
  File "/Users/elie/anaconda3/envs/phy2/lib/python3.7/site-packages/phy/cluster/supervisor.py", line 661, in __init__
    spike_clusters, spikes_per_cluster=spc, new_cluster_id=new_cluster_id)
  File "/Users/elie/anaconda3/envs/phy2/lib/python3.7/site-packages/phy/cluster/clustering.py", line 157, in __init__
    assert np.all(self._spike_clusters < self._new_cluster_id)
AssertionError`

rossant, 2019-11-13:
could you delete the `.phy` subfolder in your data directory and try again (or, equivalently, use the `--clear-cache` option of the command-line in the latest version of phy)

julieelie, 2019-11-13:
Yes! I finally found about this hidden folder yesterday and indeed deleting that fixed the problem. Good to know that data are stored here.
Thanks!

closed 2019-11-13 by julieelie