cortex-lab/phy #917: index error

catubc, 2019-11-18:
Hi 
I am getting a crash similar to an older issue ( IndexError for Viewing Clusters #790 ):

```
(phy) cat@cat-work:/media/cat/2TB/liam/49channels/data1_allset/phy$ phy template-gui params.py
Attribute Qt::AA_ShareOpenGLContexts must be set before QCoreApplication is created.
18:19:03.691 [E] __init__:62          An error has occurred (IndexError): index 5373306 is out of bounds for axis 0 with size 5373197
Traceback (most recent call last):
  File "/home/cat/anaconda3/envs/phy/bin/phy", line 11, in <module>
    load_entry_point('phy', 'console_scripts', 'phy')()
  File "/home/cat/anaconda3/envs/phy/lib/python3.7/site-packages/click/core.py", line 764, in __call__
    return self.main(*args, **kwargs)
  File "/home/cat/anaconda3/envs/phy/lib/python3.7/site-packages/click/core.py", line 717, in main
    rv = self.invoke(ctx)
  File "/home/cat/anaconda3/envs/phy/lib/python3.7/site-packages/click/core.py", line 1137, in invoke
    return _process_result(sub_ctx.command.invoke(sub_ctx))
  File "/home/cat/anaconda3/envs/phy/lib/python3.7/site-packages/click/core.py", line 956, in invoke
    return ctx.invoke(self.callback, **ctx.params)
  File "/home/cat/anaconda3/envs/phy/lib/python3.7/site-packages/click/core.py", line 555, in invoke
    return callback(*args, **kwargs)
  File "/home/cat/anaconda3/envs/phy/lib/python3.7/site-packages/click/decorators.py", line 17, in new_func
    return f(get_current_context(), *args, **kwargs)
  File "/home/cat/code/phy/phy/apps/__init__.py", line 158, in cli_template_gui
    template_gui(params_path, **kwargs)
  File "/home/cat/code/phy/phy/apps/template/gui.py", line 202, in template_gui
    gui = controller.create_gui()
  File "/home/cat/code/phy/phy/apps/base.py", line 1542, in create_gui
    self.supervisor.attach(gui)
  File "/home/cat/code/phy/phy/cluster/supervisor.py", line 956, in attach
    gui=gui, sort=gui.state.get('ClusterView', {}).get('current_sort', None))
  File "/home/cat/code/phy/phy/cluster/supervisor.py", line 777, in _create_views
    gui, data=self.cluster_info, columns=self.columns, sort=sort)
  File "/home/cat/code/phy/phy/cluster/supervisor.py", line 930, in cluster_info
    return [self._get_cluster_info(cluster_id) for cluster_id in self.clustering.cluster_ids]
  File "/home/cat/code/phy/phy/cluster/supervisor.py", line 930, in <listcomp>
    return [self._get_cluster_info(cluster_id) for cluster_id in self.clustering.cluster_ids]
  File "/home/cat/code/phy/phy/cluster/supervisor.py", line 763, in _get_cluster_info
    out[key] = func(cluster_id)
  File "/home/cat/code/phy/phy/apps/base.py", line 490, in get_cluster_amplitude
    template_id = self.get_template_for_cluster(cluster_id)
  File "/home/cat/code/phy/phy/utils/context.py", line 154, in memcached
    out = f(*args, **kwargs)
  File "/home/cat/code/phy/phy/apps/base.py", line 477, in get_template_for_cluster
    st = self.model.spike_templates[spike_ids]
IndexError: index 5373306 is out of bounds for axis 0 with size 5373197
```

rossant, 2019-11-19:
This happens even if you delete the `.phy` subfolder in the data directory and relaunch the GUI?
If so, could you send me the dataset?

catubc, 2019-11-19:
That fixed it! Thanks!

closed 2019-11-19 by catubc