cortex-lab/phy #919: waveforms in only few channels on PHY2

lalittaSA, 2019-11-21:
[image]
Hi,

I got PHY2 working now. But I don't have the waveforms properly displayed. I always got 2 channels and the other ones are flat. I tried removing ./phy folders everywhere but it didn't solve the problem.

Thanks for your help!
Lalitta

rossant, 2019-11-21:
Could you send me your dataset?

lalittaSA, 2019-11-21:
Thanks for you reply. Do you mean the whole folder with raw data? It's ~25GB.
What would be the best way to transfer it to you?

rossant, 2019-11-21:
Yes but you can not send the raw data (just give me its exact file size in bytes)
I also don't need the `.phy` subfolder
You could use dropbox or google drive?

lalittaSA, 2019-11-21:
I've compressed the folder (except the raw data) and sent it to you through DropSend (My dropbox folder doesn't have enough space).

Thanks again for your help.

rossant, 2019-12-24:
fixed in the latest version

closed 2019-12-24 by rossant

lalittaSA, 2019-12-24:
Many thanks again, Cyrille!
L