cortex-lab/phy #922: Template Gui Distorted and unusable in Windows 10

jmhasse01, 2019-11-27:
My apologies if this sort of thing has been addressed before. Running the template-gui command opens a GUI which is distorted and unusable (see images below). This machine is a laptop with 32gb RAM and an Nvidia GTX 1660. I can run the template-gui command on a desktop on the same files and it works perfectly (see images below). 

I followed the installation instructions on both machines. I'm also running Anaconda3 on both machines. I'm worried that there is a display issue with the laptop, possibly between the integrated graphics card and the dedicated graphics card. I've also included the --debug output for both cases. Any help is appreciated!

Thanks,
Mike

Distorted GUI:
[image]

Distorted GUI debug:
[image]

Good GUI:
[image]

Good GUI debug:
[image]

subhacom, 2019-12-01:
Can confirm same problem on a Windows 10 laptop with 16 GB RAM and NVIDIA GeForce  GTX 1050 Ti graphics card.

jmhasse01, 2019-12-02:
Ok, I was able to solve the distortion problem by forcing Powershell to run from the discrete graphics card in the Nvidia Control Panel. Running from the integrated graphics card causes distortions in the GUI.

closed 2019-12-03 by jmhasse01

rossant, 2019-12-04:
Thank you for letting us now @jmhasse01 
How did you force Powershell to run from the discrete GPU?

jmhasse01, 2019-12-04:
I used the Nvidia control panel: right click on Desktop>Nvidia Control Panel>Desktop> 'Add "Run with Graphics Processor" option'. Then I was able to right click on the Powershell shortcut and select which graphics card to use. 

I'm guessing there's a way to set the discrete graphics card as the default for Powershell, but I haven't figured out how to do that yet.

subhacom, 2019-12-05:
Thanks for sharing this, it solves the problem. 

After enabling the "Run with graphics processor" menu option, this appears as a submenu in the right-click pop-up menu. After expanding that submenu, it shows the list of GPUs to choose from, and then at the bottom of the list "Change default graphics processor" which is supposed to do that.

Another way I found was to go to 3D-settings in NVIDIA control panel, select "Manage 3D settings" in the menu tree on the left, and then go to "Program settings" tab. One can set the preferred GPU for individual programs here.

R-Norris, 2020-09-17:
Thank you for sharing these solutions, I ran into a similar issue with distorted or blank view windows on a Windows 10 machine and this thread has been very helpful. 

I just wanted to add that with the recent May 2020 update (20H1), there is no longer a “Run with graphics processor” option in the NVIDIA Control Panel. The solution that worked for me was to go into the Windows settings > System > Display > Graphics Settings and then use Browse to add phy and Options to select the NVIDIA GPU. This forces Windows to run phy through the selected GPU and fixed all the display issues I was having.