cortex-lab/phy #925: Reset gui view for phy2

JannaAarse, 2019-12-06:
Hello, 

I have a problem with the view of the gui. I opened it for a new dataset and the view is completely different and kind of unusable to check the quality of the clusters. I also opened an old datafile that I sorted yesterday but this one now also looks different. I attached a screenshot of it. 
I tried to reset the gui configuration after the description here 
https://phy.readthedocs.io/en/latest/installation
but it didn't changed anything, then I deleted the environment phy2 and reinstalled it but it still looks the same. 
Is there any other way to reset the look of it ? 

Best, 
Janna 

[image]

410pfeliciano, 2019-12-14:
Update your NVIDIA drivers. Phy2 has this problem with some Intel GPU's, make sure that python runs with NVIDIA graphics.

closed 2019-12-24 by rossant

tpbollu, 2019-12-31:
> Hello,
>
> I have a problem with the view of the gui. I opened it for a new dataset and the view is completely different and kind of unusable to check the quality of the clusters. I also opened an old [...]

I had the same issue, it appears to have been caused by an update to the Intel Graphics Driver. Rolling back the driver to the previous version solved the problem.

JannaAarse, 2020-01-02:
Thanks for your help, unfortunately neither of the solutions did work. There was no update on my Intel so I can not roll it back. And the update of the NVIDIA didn't changed the view. I found out that if I don't run it in full screen mode I can separate the Clusterview and at least still see all the features but when I want to close everything it automatically also closes Anaconda in total.