cortex-lab/phy #926: Unreferenced Clusters and unable to split in filter mode

bapungiri, 2019-12-09:
### Issue 1:

After upgrading to latest version of phy2 using the following commands:

```
pip install --upgrade git+https://github.com/cortex-lab/phy.git@dev
pip install --upgrade git+https://github.com/cortex-lab/phylib.git

```
I am seeing these errors in the terminal:
```

11:43:40.118 [W] model:535            Warning: unreferenced clusters found in templates !
11:43:40.415 [W] model:555            Warning: unreferenced clusters found in spike_clusters !
11:43:41.195 [W] state:183            Could not copy non-existing default state file /home/bapung/conda3/envs/phy2/lib/python3.7/site-packages/phy/apps/template/static/state.json.
11:44:15.003 [W] variable:389         Data is empty for a_box_index
11:44:15.009 [W] variable:389         Data is empty for a_hist_index
11:44:15.009 [W] variable:389         Data is empty for a_position
11:44:15.063 [W] variable:389         Data is empty for a_box_index
11:44:15.078 [W] variable:389         Data is empty for a_hist_index
11:44:15.079 [W] variable:389         Data is empty for a_position
11:44:15.080 [W] variable:389         Data is empty for a_box_index
11:44:15.080 [W] variable:389         Data is empty for a_color
11:44:15.080 [W] variable:389         Data is empty for a_position
11:44:15.113 [W] variable:389         Data is empty for a_box_index
11:44:15.128 [W] variable:389         Data is empty for a_hist_index
11:44:15.129 [W] variable:389         Data is empty for a_position
```

### Issue 2:

Splitting cluster using lasso does not work in "filter cluster" mode. I was using **group != 'noise'**.

rossant, 2019-12-17:
1) the "data is empty" messages are often caused by a window that is too small, I'll get rid of them as they can be discarded
2) is there an error message?

bapungiri, 2019-12-17:
1. Sounds good.
2. No i don't see any error message. But i do see those warnings every time I launch template-gui. Apart from that everything works fine.

closed 2019-12-24 by rossant