cortex-lab/phy #928: Display of TraceView and WaveformView for parallel probes.

JoseGuzman, 2019-12-16:
If I understand correctly, [Phy](github.com/cortex-lab/phy/) plots the **WaveformView** based on the probe file, and it takes the largest y-coordinate to plot it above and the smallest y-value to plot down (*). I have a probe file whose (x,y) going (from [0,0] to say, [35,-305]). While the geometry of the probe is preserved, the channel numbers in **TraceView** _are in the opposite direction_ than in **WaveformView** (see figure). 

[image]

When I tried to set the coordinates in positive values (in the previous example, from [0,0] to [35,305], the geometry of the probe is upward and this time, the TraceView is correct, but the WaveformView is in the opposite direction. Am I missing something?

I was wondering what is the way Phy plots **TraceView** and **WaveformView**, because due to the reasoning above (*) it will be difficult to visualize parallel probes. Is it because Phy is oriented towards linear probes?. Would it be possible to arrange parallel probes (e.g., four linear probes), and have the channels simply oriented from 1 to n downwards in the **TraceView** but keeping the geometry in **WaveformView**?

Thanks in advance!

Jose

rossant, 2019-12-17:
There should be an option in the trace view to switch the origin (view menu, trace view, switch origin), does it work?

JoseGuzman, 2019-12-17:
Yes, ALT+O. It's embarrassing that  I missed it. Thank you very much.

closed 2019-12-17 by JoseGuzman