cortex-lab/phy #930: mismatch between channel ID in Cluster/Similarity views and actual amplitude of templates/waveforms

julieelie, 2020-01-14:
Hello,
I've noticed that some of my clusters don't seem to be assigned to the correct "best channel". I first thought that the best channel was taken from the output of Kilosort2 in template_feature_ind.npy but I found that it might be calculated by Phy with _find_best_channels()

For instance here Cluster 49 is assigned to channel 2 when the waveforms amplitude are bigger on 8, 9 and 10:
[image]
and if you look at the templates, it's the same thing:
[image]

Here is another example, cluster 56, of the same dataset:
[image]
[image]

Where is that discrepancy coming from?
Thank you for your help!

rossant, 2020-01-14:
I'm not sure if this is the problem, but could you try (with the latest development version of phy and phylib) to add the following line in your `params.py`:

`show_mapped_channels = False`

This parameter controls whether the channel mapping is used or not when displaying the channel labels in the different views.

If that doesn't fix it, could you please send me your dataset so that I have a look? (I don't need the large raw data file, just the npy/tsv/params.py)

julieelie, 2020-01-14:
Argh.... I went into more troubles by trying to used a clean last version of Phy and Phylib.
I erased my environment along with ~/.phy folder 

and started from scratch with:
conda create -n phy2 python pip numpy matplotlib scipy h5py pyqt cython -y
conda activate phy2
pip install colorcet pyopengl qtconsole requests traitlets tqdm joblib click mkdocs
pip install git+https://github.com/cortex-lab/phy.git@dev
pip install git+https://github.com/cortex-lab/phylib.git

But now I can't open the gui anymore, looks like lines changed in phy/gui/qt.py, now calling PIL, which is giving me the following error although I definitely have it on in my packages (anaconda3/lib/python3.7/site-packages/PIL). Here is the whole error when running the debug option for phy:

> 14:44:45.961 [D] __init__:68          Start capturing exceptions.
> 14:44:46.179 [D] model:560            Loading spike clusters.
> 14:44:46.246 [D] model:507            No channel shank file found.
> 14:44:46.247 [D] model:627            Loading templates.
> 14:44:46.298 [D] model:654            Templates are dense.
> 14:44:46.298 [D] model:604            Loading spike waveforms.
> 14:44:46.299 [D] model:609            Skipping spike waveforms that do not exist, they will be extracted on the fly from the raw data as needed.
> 14:44:46.299 [D] model:660            Loading the whitening matrix.
> 14:44:46.300 [D] model:667            Loading the inverse of the whitening matrix.
> 14:44:46.302 [D] model:164            Loading traces at `/Volumes/server_home/users/JulieE/LMC/LMC_CoEd/logger/20190130/Logger16/spikesorted/11689_20190130_TempTetrode_0_1_2_3_4_5_6_7_8_9_10_12_13_14_15.bin`.
> 14:44:46.303 [D] model:696            Loading features.
> 14:44:46.306 [D] model:711            Features are sparse.
> 14:44:46.306 [D] model:733            Loading template features.
> 14:44:46.375 [D] model:743            Template features are sparse.
> 14:44:46.380 [D] model:442            Load `cluster_Amplitude.tsv`.
> 14:44:46.382 [D] _misc:319            Read /Volumes/server_home/users/JulieE/LMC/LMC_CoEd/logger/20190130/Logger16/spikesorted/cluster_Amplitude.tsv.
> 14:44:46.382 [D] model:442            Load `cluster_ContamPct.tsv`.
> 14:44:46.384 [D] _misc:319            Read /Volumes/server_home/users/JulieE/LMC/LMC_CoEd/logger/20190130/Logger16/spikesorted/cluster_ContamPct.tsv.
> 14:44:46.384 [D] model:442            Load `cluster_KSLabel.tsv`.
> 14:44:46.386 [D] _misc:319            Read /Volumes/server_home/users/JulieE/LMC/LMC_CoEd/logger/20190130/Logger16/spikesorted/cluster_KSLabel.tsv.
> 14:44:46.416 [D] context:100          Initialize joblib cache dir at `/Volumes/server_home/users/JulieE/LMC/LMC_CoEd/logger/20190130/Logger16/spikesorted/.phy`.
> 14:44:46.416 [D] context:101          Reducing the size of the cache if needed.
> 14:44:46.428 [D] base:101             Add filter `high_pass`.
> 14:44:46.429 [D] config:31            Load config file `/Users/elie/.phy/phy_config.py`.
> 14:44:46.429 [D] plugin:145           Loading 0 plugins.
> 14:44:46.430 [D] context:209          The file `/Volumes/server_home/users/JulieE/LMC/LMC_CoEd/logger/20190130/Logger16/spikesorted/.phy/new_cluster_id.pkl` doesn't exist.
> 14:44:48.310 [D] context:185          Save data to `/Volumes/server_home/users/JulieE/LMC/LMC_CoEd/logger/20190130/Logger16/spikesorted/.phy/spikes_per_cluster.pkl`.
> 14:44:50.108 [D] gui:440              Creating GUI.
> 14:44:50.136 [W] state:183            Could not copy non-existing default state file /Users/elie/anaconda3/envs/phy2/lib/python3.7/site-packages/phy/apps/template/static/state.json.
> GVA info: Successfully connected to the Intel plugin, offline Gen95 
> 14:44:50.337 [D] gui:696              Add view ClusterView to GUI.
> 14:44:50.458 [D] gui:696              Add view SimilarityView to GUI.
> 14:44:50.462 [D] qt:379               Saving icon `/Users/elie/anaconda3/envs/phy2/lib/python3.7/site-packages/phy/gui/static/icons/f0e2.png` using the PIL library.
> 14:44:50.464 [E] __init__:62          An error has occurred (ModuleNotFoundError): No module named 'PIL'
> Traceback (most recent call last):
>   File "/Users/elie/anaconda3/envs/phy2/bin/phy", line 8, in <module>
>     sys.exit(phycli())
>   File "/Users/elie/anaconda3/envs/phy2/lib/python3.7/site-packages/click/core.py", line 764, in __call__
>     return self.main(*args, **kwargs)
>   File "/Users/elie/anaconda3/envs/phy2/lib/python3.7/site-packages/click/core.py", line 717, in main
>     rv = self.invoke(ctx)
>   File "/Users/elie/anaconda3/envs/phy2/lib/python3.7/site-packages/click/core.py", line 1137, in invoke
>     return _process_result(sub_ctx.command.invoke(sub_ctx))
>   File "/Users/elie/anaconda3/envs/phy2/lib/python3.7/site-packages/click/core.py", line 956, in invoke
>     return ctx.invoke(self.callback, **ctx.params)
>   File "/Users/elie/anaconda3/envs/phy2/lib/python3.7/site-packages/click/core.py", line 555, in invoke
>     return callback(*args, **kwargs)
>   File "/Users/elie/anaconda3/envs/phy2/lib/python3.7/site-packages/click/decorators.py", line 17, in new_func
>     return f(get_current_context(), *args, **kwargs)
>   File "/Users/elie/anaconda3/envs/phy2/lib/python3.7/site-packages/phy/apps/__init__.py", line 158, in cli_template_gui
>     template_gui(params_path, **kwargs)
>   File "/Users/elie/anaconda3/envs/phy2/lib/python3.7/site-packages/phy/apps/template/gui.py", line 200, in template_gui
>     gui = controller.create_gui()
[... cut ...]
>   File "/Users/elie/anaconda3/envs/phy2/lib/python3.7/site-packages/phy/gui/actions.py", line 220, in _create_qaction
>     action.setIcon(_get_icon(kwargs['icon']))
>   File "/Users/elie/anaconda3/envs/phy2/lib/python3.7/site-packages/phy/gui/qt.py", line 380, in _get_icon
>     from PIL import Image, ImageDraw, ImageFont
> ModuleNotFoundError: No module named 'PIL'

Also here is the zip to the data with the channel mapping  that gave me the mismatch channels ID in the first place:
https://drive.google.com/open?id=1-PKK-vWZVOsWoJA6kBLExzTowCufzUr8
Thanks!

rhuszar, 2020-01-14:
I had to erase my environment for completely other reasons, and now have the exact same issue as @julieelie: No module named 'PIL', can't open the GUI.

rossant, 2020-01-15:
Sorry about that, I've pushed a fix, but you can also do `conda install pillow` after you've activated your environment.

julieelie, 2020-01-16:
Running `conda install pillow` did fix the issue with the last version of the code, but when trying to update the code on my laptop I now get this error with the last version of the code you pushed:

> Traceback (most recent call last):
>   File "/Users/elie/anaconda3/envs/phy2/bin/phy", line 5, in <module>
>     from phy.apps import phycli
>   File "/Users/elie/anaconda3/envs/phy2/lib/python3.7/site-packages/phy/__init__.py", line 21, in <module>
>     from .utils.config import load_master_config
>   File "/Users/elie/anaconda3/envs/phy2/lib/python3.7/site-packages/phy/utils/__init__.py", line 17, in <module>
>     from phylib.utils.color import(
> ModuleNotFoundError: No module named 'phylib.utils.color'

rossant, 2020-01-16:
There might be a version conflict with phylib, try doing `pip install git+https://github.com/cortex-lab/phylib --upgrade` after activating the phy2 conda environment.

julieelie, 2020-01-16:
updating phylib only did not fix the error. But whipping phy2 environment and reinstalling everything did! Thank you!

closed 2020-01-22 by rossant