cortex-lab/phy #931: Phy Kwik_Gui fail to open the kwik file

saman-abbaspoor, 2020-01-14:
Hello all,

I have recently installed Klusta and ran it on the test dataset. I have a .kwik file now but can't open it with the phy kwik-gui. I recieve the follwoing error. I have check different resources to find a solution but the offered solutions did not work. I have the most updated **docopt** installed. Also, **phy**, **phycontrib** and **klusta** are all installed on the same environment. I also have **klustaviewa** which does not open as well but does not give any error. It's just not responsive. I will appreciate if you can provide a solution to this problem.

phy kwik-gui hybrid_10sec.kwik
[31m00:07:23.928 [E] __init__:62          An error has occurred (TypeError): _getfullpathname: illegal type for path parameter
Traceback (most recent call last):
  File "c:\anaconda3\envs\klusta\lib\runpy.py", line 193, in _run_module_as_main
    "__main__", mod_spec)
  File "c:\anaconda3\envs\klusta\lib\runpy.py", line 85, in _run_code
    exec(code, run_globals)
  File "C:\Anaconda3\envs\klusta\Scripts\phy.exe\__main__.py", line 7, in <module>
    sys.exit(phycli())
  File "c:\anaconda3\envs\klusta\lib\site-packages\click\core.py", line 722, in __call__
    return self.main(*args, **kwargs)
  File "c:\anaconda3\envs\klusta\lib\site-packages\click\core.py", line 697, in main
    rv = self.invoke(ctx)
  File "c:\anaconda3\envs\klusta\lib\site-packages\click\core.py", line 1066, in invoke
    return _process_result(sub_ctx.command.invoke(sub_ctx))
  File "c:\anaconda3\envs\klusta\lib\site-packages\click\core.py", line 895, in invoke
    return ctx.invoke(self.callback, **ctx.params)
  File "c:\anaconda3\envs\klusta\lib\site-packages\click\core.py", line 535, in invoke
    return callback(*args, **kwargs)
  File "c:\anaconda3\envs\klusta\lib\site-packages\click\decorators.py", line 17, in new_func
    return f(get_current_context(), *args, **kwargs)
  File "c:\anaconda3\envs\klusta\lib\site-packages\phy\apps\__init__.py", line 186, in cli_kwik_gui
    kwik_gui(path, channel_group=channel_group, clustering=clustering, **kwargs)
  File "c:\anaconda3\envs\klusta\lib\site-packages\phy\apps\kwik\gui.py", line 237, in kwik_gui
    path, channel_group=channel_group, clustering=clustering, **kwargs)
  File "c:\anaconda3\envs\klusta\lib\site-packages\phy\apps\kwik\gui.py", line 104, in __init__
    super(KwikController, self).__init__(kwik_path=kwik_path, dir_path=dir_path, **kwargs)
  File "c:\anaconda3\envs\klusta\lib\site-packages\phy\apps\base.py", line 854, in __init__
    _add_log_file(Path(dir_path) / 'phy.log')
  File "c:\anaconda3\envs\klusta\lib\site-packages\phylib\__init__.py", line 65, in _add_log_file
    handler = logging.FileHandler(filename)
  File "c:\anaconda3\envs\klusta\lib\logging\__init__.py", line 1004, in __init__
    self.baseFilename = os.path.abspath(filename)
  File "c:\anaconda3\envs\klusta\lib\ntpath.py", line 535, in abspath
    path = _getfullpathname(path)
TypeError: _getfullpathname: illegal type for path parameter
[0m

(klusta) C:\Users\Saman\Documents\ec013.527.dat\crcns\hc2\ec013.527>

rossant, 2020-01-14:
Could you try, instead of installing phy in the klusta environment, to install klusta in the phy environment?

bapungiri, 2020-01-14:
It wasn't working for me as well. But after installing klusta inside phy environment it did work.

`pip install klusta`

saman-abbaspoor, 2020-01-14:
Hello, thank you for your response,
Do you mean I create a separate environment named "phy" and install
packages there?

Saman Abbaspoor
Ph.D. Student | Vanderbilt University | Psychological Sciences Department
*mobile: (615) 705-94 <(615)+238-7261>16*
*email: saman.abbaspoor@vanderbilt.com <saman.abbaspoor@vanderbilt.com>*
*skype: saman95abbaspour*

rossant, 2020-01-15:
Yes, follow the instructions to install phy, and after you've activate the environment, do "pip install klusta" to install klusta. Then the kwik GUI should be available in that environment.

saman-abbaspoor, 2020-01-16:
I have installed Phy2 from the following instructions and then installed
Klusta in that environment.

https://phy.readthedocs.io/en/latest/installation/#instructions

However, the problem still persists. Here is the environment, and the
packages on it.

[image: image.png]

[image: image.png]

[image: image.png]

here's an image of the error message:

[image: image.png]

Are there any other potential solutions to this issue?

Saman Abbaspoor
Ph.D. Student | Vanderbilt University | Psychological Sciences Department
*mobile: (615) 705-94 <(615)+238-7261>16*
*email: saman.abbaspoor@vanderbilt.com <saman.abbaspoor@vanderbilt.com>*
*skype: saman95abbaspour*

rossant, 2020-01-20:
Could you activate the phy2 environment, do `conda install pillow`, and try again?

closed 2020-06-08 by rossant

mchini, 2020-06-11:
Hello Cyrille,
I am having an error similar to the one described in the first post of this thread, despite having klusta installed in the phy environment.
Any suggestions on how to solve the issue?
Thanks!

[image]

rossant, 2020-06-11:
As a quick test could you open `c:\users\mchini\appdata\local\continuum\anaconda3\envs\phy2\lib\site-packages\phylib\__init__.py` in a text editor, go line 65, and replace
`handler = logging.FileHandler(filename)`
by
`handler = logging.FileHandler(str(filename))`
? Curious to know if that's the problem.

mchini, 2020-06-11:
Thanks for the fast answer! 
The error now changed to this:
[image]

rossant, 2020-06-11:
`c:\users\mchini\appdata\local\continuum\anaconda3\envs\phy2\lib\site-packages\phy\apps\kwik\gui.py` line 44, replace
`shutil.copy(path, path_backup)`
by
`shutil.copy(str(path), str(path_backup))`

mchini, 2020-06-11:
things are moving, but we're not quite there yet :)
[image]

rossant, 2020-06-11:
`c:\users\mchini\appdata\local\continuum\anaconda3\envs\phy2\lib\site-packages\klusta\kwik\h5.py` line 95 try replacing `self.filename` by `str(self.filename)`

mchini, 2020-06-11:
[image]

rossant, 2020-06-11:
The HDF5 library apparently cannot open the kwik file, could you send it to me via dropbox or similar?

mchini, 2020-06-11:
I wrote you on your gmail account. Thanks again for your help. Much appreciated to say the least