cortex-lab/phy #932: bug on similarity view

rtomioka, 2020-01-14:
Hi,

We recently started to use Kilosort&phy to do clustering units.
Although it might be the major bug, I just want to know whether the bug is not serious.

We found the bug on similarity view.
When I chose cluster133, I could find cluster 2 on the similarity view.
[image]

But when I chose cluster137, I missed cluster 2 on the similarity view.
[image]

Please let me know whether it is major problem or not.
Best,
Ryohei

rossant, 2020-01-14:
I suggest you sort the similarity views by decreasing similarity, by clicking on the "similarity" column header.

Also you should know that only the most similar clusters are shown in the similarity view, so in general you cannot expect the displayed similarity matrix to be symmetric.

rtomioka, 2020-01-14:
Hi,

Thank you for your quick reply.
I got your idea.

Best,
Ryohei

closed 2020-01-14 by rtomioka