cortex-lab/phy #934: Can't open GUI, KeyError: 'artifacts'

rhuszar, 2020-01-15:
For various reasons, I had to reinstall Anaconda and Phy2 recently, but I've been having issues running the GUI ever since. This is the error message I am getting when running in debug mode:

```
(phy2) C:\Users\Roman\Documents\DATA\e15_4m2_191213\Kilosort_2019-12-17_004941>phy template-gui params.py --debug
[90m11:08:37.100 [D] __init__:68          Start capturing exceptions.[0m
[33m11:08:38.027 [W] model:545            Unreferenced clusters found in templates (generally not a problem)[0m
[90m11:08:38.027 [D] model:560            Loading spike clusters.[0m
[33m11:08:38.346 [W] model:567            Unreferenced clusters found in spike_clusters (generally not a problem)[0m
[90m11:08:38.348 [D] model:627            Loading templates.[0m
[90m11:08:38.348 [D] model:654            Templates are dense.[0m
[90m11:08:38.348 [D] model:604            Loading spike waveforms.[0m
[90m11:08:38.348 [D] model:609            Skipping spike waveforms that do not exist, they will be extracted on the fly from the raw data as needed.[0m
[90m11:08:38.353 [D] model:660            Loading the whitening matrix.[0m
[90m11:08:38.354 [D] model:667            Loading the inverse of the whitening matrix.[0m
[90m11:08:38.356 [D] model:164            Loading traces at `C:\Users\Roman\Documents\DATA\e15_4m2_191213\e15_4m2_191213.dat`.[0m
[90m11:08:38.358 [D] model:696            Loading features.[0m
[90m11:08:38.358 [D] model:711            Features are sparse.[0m
[90m11:08:38.358 [D] model:733            Loading template features.[0m
[90m11:08:38.358 [D] model:743            Template features are sparse.[0m
[90m11:08:38.358 [D] model:442            Load `cluster_group.tsv`.[0m
[90m11:08:38.358 [D] _misc:319            Read C:\Users\Roman\Documents\DATA\e15_4m2_191213\Kilosort_2019-12-17_004941\cluster_group.tsv.[0m
[90m11:08:38.406 [D] context:100          Initialize joblib cache dir at `C:\Users\Roman\Documents\DATA\e15_4m2_191213\Kilosort_2019-12-17_004941\.phy`.[0m
[90m11:08:38.406 [D] context:101          Reducing the size of the cache if needed.[0m
[90m11:08:38.406 [D] base:101             Add filter `high_pass`.[0m
[90m11:08:38.406 [D] config:31            Load config file `C:\Users\Roman\.phy\phy_config.py`.[0m
[90m11:08:38.406 [D] plugin:145           Loading 0 plugins.[0m
[90m11:08:38.417 [D] context:209          The file `C:\Users\Roman\Documents\DATA\e15_4m2_191213\Kilosort_2019-12-17_004941\.phy\new_cluster_id.pkl` doesn't exist.[0m
[90m11:08:38.676 [D] context:185          Save data to `C:\Users\Roman\Documents\DATA\e15_4m2_191213\Kilosort_2019-12-17_004941\.phy\spikes_per_cluster.pkl`.[0m
[90m11:08:38.807 [D] gui:440              Creating GUI.[0m
[33m11:08:38.815 [W] state:183            Could not copy non-existing default state file c:\users\roman\anaconda3\envs\phy2\lib\site-packages\phy\apps\template\static\state.json.[0m
[90m11:08:39.608 [D] gui:696              Add view ClusterView to GUI.[0m
[90m11:08:39.791 [D] gui:696              Add view SimilarityView to GUI.[0m
[90m11:08:39.904 [D] gui:696              Add view WaveformView to GUI.[0m
[31m11:08:39.914 [E] __init__:62          An error has occurred (KeyError): 'artifacts'
Traceback (most recent call last):
  File "c:\users\roman\anaconda3\envs\phy2\lib\runpy.py", line 193, in _run_module_as_main
    "__main__", mod_spec)
  File "c:\users\roman\anaconda3\envs\phy2\lib\runpy.py", line 85, in _run_code
    exec(code, run_globals)
  File "C:\Users\Roman\Anaconda3\envs\phy2\Scripts\phy.exe\__main__.py", line 7, in <module>
    sys.exit(phycli())
  File "c:\users\roman\anaconda3\envs\phy2\lib\site-packages\click\core.py", line 764, in __call__
    return self.main(*args, **kwargs)
  File "c:\users\roman\anaconda3\envs\phy2\lib\site-packages\click\core.py", line 717, in main
    rv = self.invoke(ctx)
  File "c:\users\roman\anaconda3\envs\phy2\lib\site-packages\click\core.py", line 1137, in invoke
    return _process_result(sub_ctx.command.invoke(sub_ctx))
  File "c:\users\roman\anaconda3\envs\phy2\lib\site-packages\click\core.py", line 956, in invoke
    return ctx.invoke(self.callback, **ctx.params)
  File "c:\users\roman\anaconda3\envs\phy2\lib\site-packages\click\core.py", line 555, in invoke
    return callback(*args, **kwargs)
  File "c:\users\roman\anaconda3\envs\phy2\lib\site-packages\click\decorators.py", line 17, in new_func
    return f(get_current_context(), *args, **kwargs)
  File "c:\users\roman\anaconda3\envs\phy2\lib\site-packages\phy\apps\__init__.py", line 158, in cli_template_gui
    template_gui(params_path, **kwargs)
  File "c:\users\roman\anaconda3\envs\phy2\lib\site-packages\phy\apps\template\gui.py", line 200, in template_gui
    gui = controller.create_gui()
  File "c:\users\roman\anaconda3\envs\phy2\lib\site-packages\phy\apps\base.py", line 1643, in create_gui
    gui.create_views()
  File "c:\users\roman\anaconda3\envs\phy2\lib\site-packages\phy\gui\gui.py", line 676, in create_views
    self.create_and_add_view(view_name)
  File "c:\users\roman\anaconda3\envs\phy2\lib\site-packages\phy\gui\gui.py", line 654, in create_and_add_view
    view.attach(self)
  File "c:\users\roman\anaconda3\envs\phy2\lib\site-packages\phy\cluster\views\waveform.py", line 356, in attach
    super(WaveformView, self).attach(gui)
  File "c:\users\roman\anaconda3\envs\phy2\lib\site-packages\phy\cluster\views\base.py", line 497, in attach
    super(ScalingMixin, self).attach(gui)
  File "c:\users\roman\anaconda3\envs\phy2\lib\site-packages\phy\cluster\views\base.py", line 304, in attach
    gui.add_view(self, position=self._default_position)
  File "c:\users\roman\anaconda3\envs\phy2\lib\site-packages\phy\gui\gui.py", line 720, in add_view
    emit('add_view', self, view)
  File "c:\users\roman\anaconda3\envs\phy2\lib\site-packages\phylib\utils\event.py", line 141, in emit
    res.append(f(sender, *args, **kwargs))
  File "c:\users\roman\anaconda3\envs\phy2\lib\site-packages\phy\apps\base.py", line 1617, in on_add_view
    self._add_default_color_schemes(view)
  File "c:\users\roman\anaconda3\envs\phy2\lib\site-packages\phy\apps\base.py", line 1576, in _add_default_color_schemes
    colormap=colormap, categorical=categorical, logarithmic=logarithmic)
  File "c:\users\roman\anaconda3\envs\phy2\lib\site-packages\phy\cluster\views\base.py", line 162, in add_color_scheme
    colormap=colormap, categorical=categorical, logarithmic=logarithmic)
  File "c:\users\roman\anaconda3\envs\phy2\lib\site-packages\phy\utils\color.py", line 237, in __init__
    fun=fun, colormap=colormap, categorical=categorical, logarithmic=logarithmic)
  File "c:\users\roman\anaconda3\envs\phy2\lib\site-packages\phy\utils\color.py", line 264, in set_color_mapping
    self.set_cluster_ids(self.cluster_ids)
  File "c:\users\roman\anaconda3\envs\phy2\lib\site-packages\phy\utils\color.py", line 269, in set_cluster_ids
    values = self.get_values(self.cluster_ids)
  File "c:\users\roman\anaconda3\envs\phy2\lib\site-packages\phy\utils\color.py", line 318, in get_values
    values = [self._get_cluster_value(cluster_id) for cluster_id in cluster_ids]
  File "c:\users\roman\anaconda3\envs\phy2\lib\site-packages\phy\utils\color.py", line 318, in <listcomp>
    values = [self._get_cluster_value(cluster_id) for cluster_id in cluster_ids]
  File "c:\users\roman\anaconda3\envs\phy2\lib\site-packages\phy\utils\color.py", line 304, in _get_cluster_value
    return self._fun(cluster_id) if hasattr(self._fun, '__call__') else self._fun or 0
  File "c:\users\roman\anaconda3\envs\phy2\lib\site-packages\phy\apps\base.py", line 1562, in group_index
    return group_colors[group]
KeyError: 'artifacts'
```
Thanks in advance for any advice!

rossant, 2020-01-15:
can you try to append the  `--clear-state --clear-cache`  options to the `phy template-gui...` command?

rhuszar, 2020-01-15:
I just tried and got the same error.

rossant, 2020-01-15:
okay this was due to custom groups that were not properly taken into account. I just pushed a hot fix, can you update phy from GitHub and try again? (`pip install git+https://github.com/cortex-lab/phy@dev --upgrade`)

rhuszar, 2020-01-15:
That worked! Thanks so much for the prompt help with this.

closed 2020-01-15 by rhuszar