cortex-lab/phy #937: Regarding timestamp management from Neuralynx data

Prime5222, 2020-01-17:
Hi everyone. Recently we've been making some tests using automated klustering (using KLUSTA) and visualizing the results with Phy. Our data was recorded from a (Buzsaki32_32_21mm) and all .ncs files were converted to a .dat file for clustering. 

After clustering and curating using Phy, we're trying to find a way to match the timestamps recorded on Neuralynx, but I haven't been able to successfully extract this information or to find a way to match spikes to the recorded timestamps.

Any suggestions or pointers to solve this issue are greatly appreciated, and apologies if this is not the right place to ask.

All the best

rossant, 2020-01-20:
Hello, could you clarify what the neuralyx timestamps are exactly, what do they represent, how they are recorded, and in which file format?

angelonc, 2020-04-10:
I've done this successfully for Kilosort2 using Neuralynx data. I've never used KLUSTA, but  as far as I can tell, phy tells time using the input sampling rate (ie. each time point = sample number / sample rate).

If I remember correctly, to convert NLX timestamps to true time, you just need to subtract the timestamp of the recording start from all your spike times, and divide by 1e6 or so (to convert to seconds). Then for phy spike times, you just divide each 'spike time' (sample number), by the sampling rates to get them in seconds also