cortex-lab/phy #938: mismatch between channel and peak electrode

yger, 2020-01-17:
Several users (and we also noticed it) reported that in latest phy version, there seems to be a mismatch between the channel where the template is peaking the most, and the number in the ClusterView. See for example the picture below, where the mismatch is visible since the min of the template is not on channel 0.

[image]

rossant, 2020-01-17:
Does adding `show_mapped_channels = False` in the params.py fixes it? The latest version takes the channel mapping into account when displaying channel labels. I don't know if that's the problem here.

The problem is that there's an ambiguity with the channel mapping: what number should be displayed in phy, the channel label before remapping, or after? This has always been confusing and it seems like different people using different software and/or hardware don't all want to see the same thing. Which is why I added this new parameter for the time being...

yger, 2020-01-17:
Ok, but this does not seem to solve my issue here, and the mapping is fairly simple on this dummy example: only 30 channels, from 0 to 29, no twisted remapping involved. I'll double check

rossant, 2020-01-17:
okay in this case could you send me the dataset (no need for the raw data)?

yger, 2020-01-17:
Sure. Here it is. Maybe I'm missing something :-(.

[silico_0.GUI.zip](https://github.com/cortex-lab/phy/files/4077049/silico_0.GUI.zip)

rossant, 2020-01-20:
Thanks, there was indeed a bug... Fix pushed on phylib and phy, could you try again?

rossant, 2020-01-22:
closing for now, reopen if that didn't fix it for you!

closed 2020-01-22 by rossant