cortex-lab/phy #939: Yellow rectangle in amplitude view

balefebvre, 2020-01-17:
Hello,
I have not noticed it immediately but since my latest update I always see a vertical yellow rectangle in the middle of the amplitude view:
[image]
This is only an esthetic issue. Do you have an idea how to make it disappear?

```
phy 2.0a1
phylib 2.2
colorcet 2.0.2
PyOpenGL 3.1.0
requests 2.22.0
traitlets 4.3.3
qtconsole 4.5.5
tqdm 4.36.1
joblib 0.14.0
Click 7.0
mtscomp 1.0.1
mkdocs 1.0.4
```

rossant, 2020-01-17:
This rectangle should show the portion of the data that is currently shown in the trace view. You can alt+click in the amplitude view to move to that time in the trace view.

balefebvre, 2020-01-17:
I didn't know that. Nice feature! Thanks.

closed 2020-01-17 by balefebvre