cortex-lab/phy #940: Frequent (but random) segmentation fault upon performing an operation on a cluster

sebromano, 2020-01-17:
Hi,
Today I upgraded to phy2, and I am getting quite frequently (and randomly) segmentation fault errors. 
The program works fine, but now and then, when I perform an operation on a cluster, like cutting, merging and so on, the program crashes and the only error message I get looks like this: 

`18:38:38.235 [I] supervisor:699       Assigned 3402 spikes.
Segmentation fault (core dumped)
`
I would like to stress that it happens at random (at least I still couldn't find a pattern) since I can cut a cluster with no problem, but then crash after cutting the next one...

I am running Ubuntu 18.04.3. I was using phy with no problems. I guess it could be a conflict between phy and phy2. I tried to wipe out phy before installing, but maybe I failed. What would be the correcting procedure to wipe it out?

Thanks and keep up the great work, phy2 looks great!

rossant, 2020-01-17:
This is strange, I think the only possible sources of segfault are PyQt and OpenGL. Does this happen on all datasets? Would you be able to pin down the exact git commit that introduced this problem?

sebromano, 2020-01-17:
Thanks for the quick reply!
I'll try it with other datasets and come back to you with an answer.
I installed phy2 for the first time today following the[ instructions](https://phy.readthedocs.io/en/latest/installation/), previously I was using phy, so I couldn't say which phy2 commit introduced the problem.

sebromano, 2020-01-21:
Hi Cyrille,
it actually might be related to the particular dataset I was working on. I'll have to continue experimenting  to be sure, but I'll let you know.
Thanks!

closed 2020-06-08 by rossant