cortex-lab/phy #941: FeatureView Ctrl+/- functionality stops working

dervinism, 2020-01-20:
I noticed that FeatureView stop working properly soon after starting Phy. In particular, Ctrl+ and Ctrl- would affect the zoom of data points in grey but not the ones appearing in colour. This affects both dev and regular versions.

rossant, 2020-01-20:
I can't reproduce the problem. What is your OS? Do you have the same problem if you run phy with the `--clear-state` and `--clear-cache` options from the command line?

dervinism, 2020-01-20:
I'm using Windows 10. It works fine again after restarting Phy but quickly the same problem occurs. However, now I noticed that by Ctrl clicking a channel in the WaveformView fixes the issue with the FeatureView.

rossant, 2020-01-20:
any error message in the terminal?

dervinism, 2020-01-20:
No error messages with --debug

rossant, 2020-01-22:
to reopen if reproduced

closed 2020-01-22 by rossant