cortex-lab/phy #942: Mismatch between amplitudes in the WaveformView and AmplitudeView

dervinism, 2020-01-20:
There often seems to be a mismatch between amplitudes in the WaveformView and AmplitudeView.
[image]

rossant, 2020-01-20:
You see the same thing if you switch amplitude type by pressing "a" several times?

dervinism, 2020-01-20:
The AmplitudeView changes. I can see that the scale changes as well. What does pressing "a" do? Does it flip between different recording channels?

dervinism, 2020-01-20:
Oh, I see it is indicated in the bottom left corner. So the maximum recording channel is based on the blue waveform, I pressume.

rossant, 2020-01-20:
yes, see https://phy.readthedocs.io/en/latest/visualization/#different-types-of-amplitudes

dervinism, 2020-01-20:
And I can change the channel by Ctrl clicking. That's good to know. Thanks

closed 2020-01-22 by rossant