cortex-lab/phy #944: Slow AmplitudeView for large dataset

bramn22, 2020-01-24:
Hi,

The AmplitudeView is extremely slow (often more than one minute for a cluster) when analyzing sorted NeuroPixel data.
All other views load fast and work without any problems. We used Phy1 previously, and there was no issue.

In the data, there are 496 identified clusters and in total around 5.000.000 spikes. The number of spikes within the selected cluster does not seem to greatly influence the wait time.

rossant, 2020-01-24:
Is the `raw` amplitude type selected in the amplitude view? You can change by pressing `a`. This mode is extremely slow at the moment as every waveforms needs to be fetched on demand.

bramn22, 2020-01-24:
That worked, thank you!

closed 2020-01-24 by bramn22