cortex-lab/phy #945: Blank views in phy2 launched via windows remote desktop

tkskdm1, 2020-01-24:
Hello friends,

I am trying to use phy (phy2) via Microsoft remote desktop; Windows10 runs a host, and a client is in macOS (10.14). In this setting, phy can be launched normally but all the views are left blank. No error message emerged in the prompt, but, when the phy GUI closed, the following showed up in the prompt;
> QWaitCondition: Destroyed while threads are still waiting

Chrome Remote Desktop could start and work with phy without the problem. 

I found that this problem can be evaded by starting phy via Chrome Remote Desktop or directly on a host machine. Once phy starts normally, it can be remotely controlled via Microsoft remote desktop application. In other words, this problem occurs specifically when phy is launched via Microsoft remote desktop.

Is there any way to launch phy straight via Microsoft remote desktop with fully functional views?

rossant, 2020-01-24:
This may be due to the remote desktop software not fully supporting OpenGL, which phy uses. This is a known long-standing issue with OpenGL and remote desktop software, and I'm not sure there is a better fix than the one you mention at the moment.

tkskdm1, 2020-01-24:
Thank you for the quick reply, Cyrille. I got the same impression after several hours of web survey...

closed 2020-02-07 by rossant

joramvanrheede, 2021-05-05:
NVIDIA have now published a tool for GPU acceleration of Windows remote desktop - Does this remove the need for a workaround?
https://www.khronos.org/news/permalink/nvidia-provides-opengl-accelerated-remote-desktop-for-geforce-5e88fc2035e342.98417181
If this works, it would also be relevant to #737

joramvanrheede, 2021-05-05:
Windows Remote Desktop now does seem to support OpenGL, but using the GPU rendering seems disabled by default in a group policy. You can enable it - this fixed it for me, see link for instructions:
https://community.esri.com/t5/implementing-arcgis-questions/enabling-gpu-rendering-on-windows-server-2016-windows-10-rdp/td-p/658522