cortex-lab/phy #946: GUI issue in Windows 10

Tom-2266, 2020-01-28:
I am having a problem trying to open a data set. The interface opens with huge font and is basically unusable. See attached picture:
[image]

I have seen that other people have had the same problem but I have been unable to fix it. The graphics card I have is Intel UHD Graphics 630. I have tried updating drivers and rolling back to previous driver versions. I don't have a dedicated graphics card. 
Any help would be very much appreciated

rossant, 2020-01-28:
Unfortunately I don't know how to fix this without a non-intel graphics chipset, and I'm unable to reproduce this at the moment.

It might be helpful if you could try to close all views in the GUI, and see if you still have this problem with just 1 or 2 views open.

Tom-2266, 2020-02-06:
Hi, sorry for the long reply.
I can close views and move them around outside the template GUI window and they look fine
I can also add some views to the template GUI window and it seems to work
[image]

But if i try to add in the cluster view or similarity view it completely messes up the whole display
[image]
 
So it seems to be an issue with those two views. Strangely they both seem to work well on their own or together but the problems arise when adding in another view

rossant, 2020-02-06:
Thanks, that was helpful. I'll try to tackle this after the next release.

rossant, 2020-02-07:
Closing, discussion now on https://github.com/cortex-lab/phy/issues/951

closed 2020-02-07 by rossant