cortex-lab/phy #947: WaveformView issue while comparing clusters

iinnpp, 2020-01-30:
Hi! I am using kwik-gui and have encountered an issue in WaveformView while comparing clusters. 

WaveformView works for one cluster:
[image]
[image]

But when I compare clusters and select cluster 3 in ClusterView and cluster 20 in SimilarityView the issue popes up:
[image]
So now in cluster 20 red traces are all over the channels even though it should be red only in channel 2 and 3.

Today I have installed latest phy version and the issue is present. It is also reproducible on 2 more PCs and with different .kwik files.

System info: 
phy --version outputs  phy, version 2.0a1
Windows 10 Education ver1909 18363.535

jvschweim, 2020-01-30:
I also have an issue with waveforms today, when I scroll through the cluster data, the waveform view does not update from one cluster to the next, but only randomly. I use the same phy version and also Windows 10...

rossant, 2020-01-30:
@iinnpp thanks for the bug report, it will be fixed in the next release

@jvschweim this is a different issue. Do you scroll through the clusters very quickly? For performance reasons, there is a maximum speed at which you can scroll through the clusters while having the views updating at every selection changes. You may play with this by disabling temporarily the updates of other views, by clicking on the "refresh" little button at the top right of every view. If you deselect this, the view won't update at every selection change and the GUI will be more responsive.

iinnpp, 2020-01-30:
@rossant thank you for a quick response!

jvschweim, 2020-01-30:
@rossant thanks for the quick reply. I give it several seconds and the waveform cluster doesn't update. Only after I click onto the 5th or 6th cluster the waveform changes. However when I close phy I get an error message with the QWaitCondition.

I have also closed all the other views just to check, and only the WaveformView is affected. And it also doesn't update when I all other views are closed. This seems to be a new bug, as I haven't had this issue when I used phy2 last in December.

jvschweim, 2020-01-30:
@rossant it's sorted now. sorry about this, I deleted the memory in the .phy folder, and now it seems to work again.

closed 2020-02-07 by rossant