cortex-lab/phy #949: incorrect waveform view and probe view

rajatsaxena, 2020-02-04:
I am seeing incorrect waveform view and probe view when I load the kilosort output data. 

[image]

This is the channel map scatter plot generated using `channel_positions.npy`. This map matches our input channel map to the kilosort.
[image]

I have tried deleting the `.phy` folder and refreshing the views but nothing. This problem exists surprisingly for only this specific dataset. Other datasets with different channel map load without any issues.

rossant, 2020-02-04:
Does the problem still occur whether you add `show_mapped_channels = False` or `True` in your `params.py`, and delete the `.phy` subdirectory, and if you use the latest version of phy and phylib on GitHub?

rajatsaxena, 2020-02-04:
updating the version fixed the issue but now switching to different clusters leads to display of incorrect waveforms and crashing of phy without any error message

rossant, 2020-02-05:
Okay then could you send me your dataset? If there is an issue with the raw waveforms, I will need the raw data as well. Thanks.

rajatsaxena, 2020-02-05:
Hi, I have added all the data here:
https://drive.google.com/drive/folders/1-JRdNN_A2GkB11iX89HhTYreVxjIwqXy?usp=sharing

**Channel map**
[image]

**Phy output**
[image]

rajatsaxena, 2020-02-07:
I just redid all the installation after downloading the repository rather `pip install` and everything seems to be back to normal. Thank you so much for your time.

closed 2020-02-07 by rajatsaxena