cortex-lab/phy #950: 'phy' is not recognized as an internal or external command, operable program or batch file

RynzzZ, 2020-02-06:
Hi,

Sorry to bother you guys. 

I kept getting this error message - 'phy' is not recognized as an internal or external command, operable program or batch file - when running the 'phy template-gui params.py' command. 
I'm running phy on my PC (win10, Anaconda 2019.03, Python 3.7, GPU GTX 1070).

I strictly followed the installation instruction on the wiki (create phy2 env and pip install), and every step seemed successful and I didn't get any error messages, but everytime 'phy' was not recognized. 

Here is the list of packages installed in phy2 environment:
[image]

I looked into these two issues (#924 and #855 ), but their solution didn't work in my case - I have confirmed that git was in the path, and seemed that all packages needed were installed. Do they need to be a specific version? I just installed the latest version... 

One thing that I found wierd is that when I ran the pip install for phy and phylib with the phy2 env activated, conda was able to successfully build phy and phylib, but conda installed them directly in the C:\Anaconda3\envs\Lib\site-packages, instead of C:\Anaconda3\envs\phy2\Lib. Is this why conda cannot find phy as an command when I call it in the phy2 env? If so, how can I fix it? 

On the contrary, I was able to run phy2 on my personal mac book pro by simply follwing the same instruction provided on the wiki website. I don't know what's wrong with my lab PC and why it's different from my mac. 

I have been working on this bug for several days and made no progress...It is deeply appreciated if anyone can help with this issue. Big thanks in advance!!!

Qianyun

rossant, 2020-02-06:
This is really strange. If you do "pip install" after activating the conda environment, and if "pip" is installed in that environment (which it seems to be), then it should install it in the environment.
An alternative would be to install phy in the root environment, so you'd need to install the dependencies in the root, as follows:

```
conda deactivate
conda install numpy matplotlib scipy h5py pyqt cython pillow
pip install colorcet pyopengl qtconsole requests traitlets tqdm joblib click mkdocs
pip install git+https://github.com/cortex-lab/phy.git@dev
pip install git+https://github.com/cortex-lab/phylib.git
```

RynzzZ, 2020-02-06:
> This is really strange. If you do "pip install" after activating the conda environment, and if "pip" is installed in that environment (which it seems to be), then it should install it in the [...]

Thanks for your advice! Yes, installing it in the root env did solve the problem. Thanks a lot!

closed 2020-02-06 by RynzzZ