cortex-lab/phy #953: spikes after the end of the recording?

scally12, 2020-02-11:
I am new to kilosort and phy2. I am going to change spike sorting program to kilosort from the Plexon offsorter.
My recoding system is 128 channels TDT system. First, I sorted with kilosort2 and tried to use phy2 to check the sorting. When I run phy2, I see as 
[W] model:395 There are 9850901/18195593 spikes after the end of the recording.
I am wondering whether it is an error of reading the rawdata or something else.
Could you help me out, please?

scally12, 2020-02-11:
I got it. It's related to the raw data. Thanks.

closed 2020-02-11 by scally12

lachioma, 2021-01-26:
> I got it. It's related to the raw data. Thanks.

Could you please explain, I am also wondering...
Thank you!

ensorpalacios, 2022-03-02:
I run KS3 and everything alright. I then run KS2 on same dataset and now whenever I try to visualise KS3 data with phy I get this error too. Any idea why?