cortex-lab/phy #955: TraceView

manfredjoswald, 2020-02-14:
I am a relatively new user and have this week upgraded to the latest phy2 beta version. Now I notice many bare stretches of my signal (see attached image of the TraceView window) and I wonder now if there is a problem with my KiloSort processing steps or if there is another explanation for this type of signal? 

The data was acquired with 64-channel silica probes in the prefrontal cortex.

[image]

rossant, 2020-02-18:
This may not be related to phy -- have you ever seen this problem before with your data and earlier versions of phy, or other software?

manfredjoswald, 2020-02-18:
No I could not use the trace view feature in the earlier version of phy that I was using before. I agree it appears to be a problem with the pre-processing of my data. I was careful not to saturate any channels during acquisition. The original data is fine and I have already analysed parts of it extensively in Matlab. Thank you for the comment!