cortex-lab/phy #957: GUI opening up incorrectly on Windows (even with upgrade)

LongLabGit, 2020-02-18:
[image]

rossant, 2020-02-18:
This is a known issue on Windows with Intel graphics drivers. Do you have another graphics card you could use?

LongLabGit, 2020-02-18:
No I don't. Is there any other workaround? Also, is this occurring because of the new update to phy2?

rossant, 2020-02-19:
Unfortunately I don't know another workaround at the moment... This is effectively related to the upgrade to PyQt5 in phy2

Charles-Elliott, 2020-03-03:
I've had the same problem.
My solution was to detach all windows from the main interface, that way they get their normal scaling back and you can work normally. Hope it helps!

rossant, 2020-03-03:
@Charles-Elliott do you mean that the problem disappears when all windows are in floating mode, and the problem reappears if one (or all?) of the windows gets redocked again to the main window?

Charles-Elliott, 2020-03-03:
Yes, basically. Strangely enough, it's fine when 1 or 2 windows are docked, but any more than that make the gui glitch as shown in the picture.

rossant, 2020-03-03:
Does the problem occurs when both the cluster view and similarity view are floating, and several other views are docked?

Charles-Elliott, 2020-03-03:
The problem occurs when SimilarityView and/or ClusterView are docked together **with any other view**.
Similarity and Cluster together are fine, so is any combination of other views together (I tested with about 8 different views and could rearrange and re-size them without issue).

rossant, 2020-03-04:
OK thanks, so the problem seems to occur as soon as an HTML table view is docked, when using a Windows Intel driver.

ArchmageIlmryn, 2020-03-04:
How does one detach the windows while it is this glitchy? I have a similar issue (using the phy gui for spyking circus), but don't see any way to detach windows.

[image]

Charles-Elliott, 2020-03-04:
It's a bit tricky: you do it as you normally would by click and drag, but you need to locate where the window's title bar (or whatever the bar you drag it with is called) is.
This bug affects only the gui, so the actual window is still the size it normally should be. With a few attempts, you should manage. Go for the cluster and similarity views first as they seem to cause the issue.

Hope it's clear enough; tell me if I'm wrong!

LongLabGit, 2020-03-04:
Ok just bought a fancy non-Intel graphics card. I am able to pull out the windows now, but now am still running into many issues. If I click on one unit, slowly things pull up BUT for any subsequent unit I click, none of the viewing windows update themselves  (I have plenty of useable RAM, so it's not that).

beholey, 2020-03-04:
Also having these problems: distorted graphics on the gui when uploading my data, distortion clears up when I undock nearly all the windows, but then when I switch units the other windows do not update. I thought it might be a problem with a large dataset but Im having issues with a small one as well. This small one had been previously sorted without a problem before I updated phy2. We do have a high computing power computer in lab with 36GB RAM and a GPU (NIVDIA I think), and I have no problem running my data on this one. These isnt a feasible long term solution though

rossant, 2020-03-04:
some interesting links
https://discourse.vtk.org/t/intel-uhd-620-vtk-qt-bug-with-new-drivers-on-windows/1694/11
https://gitlab.kitware.com/vtk/vtk/issues/17727
https://gitlab.kitware.com/paraview/paraview/issues/19364

it looks like installing the Intel driver version 26.20.100.7755 from the Intel website might fix it? Could someone try it?

ArchmageIlmryn, 2020-03-05:
Updating to the latest Intel driver available through the hp website (26.20.100.7263)(hp computers apparently require the hp-specific driver) did not help.

Detaching the windows does seem to work though.

rossant, 2020-03-05:
I guess drivers older than 26.20.100.7755 are not going to work -- hoping the hp-specific drivers will be updated soon enough.

LongLabGit, 2020-03-05:
This is a brand new, non Intel driver.

rossant, 2020-03-05:
@LongLabGit you confirm that 1) the screen is plugged to the non-Intel GPU and 2) the Intel driver is *not* running in the phy process (there's sometimes utility software that lets you choose which GPU to run on each process)?

LongLabGit, 2020-03-12:
Thank you, it seems to be working now as it runs on the new GPU!

closed 2020-03-12 by LongLabGit

rly, 2020-04-05:
> it looks like installing the Intel driver version 26.20.100.7755 from the Intel website might fix it? Could someone try it?

@rossant I am on Windows with Intel UHD Graphics 620 and no GPU. I had a slightly older graphics driver and had the same GUI issues that others reported. I updated the graphics driver to the latest available on my system: 26.20.100.7810 and that fixed the issue. Thanks!

Prior to updating the driver, undocking the Cluster View and Similarity View seemed to resolve the issue, but this was tricky to do. I did it programmatically by running phy from the latest source code and adding the following lines to `apps/template/gui.py` between `gui.show()` and `run_app()` at line 200:
```python
from phy.cluster import ClusterView, SimilarityView
gui.get_view(ClusterView).dock.setFloating(True)
gui.get_view(SimilarityView).dock.setFloating(True)
```
This might help any folks who are stuck with an old driver and have trouble undocking the views.

rossant, 2020-04-06:
Thanks @rly very helpful! I'll put a link to your comment in the documentation.