cortex-lab/phy #958: Cluster / Similarity view

Charles-Elliott, 2020-02-21:
A small suggestion:
It would be nice to be able to customise the cluster and similarity views, especially being able to shuffle around what is displayed.
Currently, I have to scroll back and forth constantly to see both cluster id, depth and KsLabel for instance (I have just the one screen and screen space is precious in phy).

Additionally, I'd like to have the option to sort clusters according to manually-given label and/or having an option to count the number of good/MUA clusters to get an idea of the quality of the data.

Otherwise, the software is pretty nice :)

rossant, 2020-02-26:
Have you tried [this plugin](https://phy.readthedocs.io/en/latest/plugins/#customizing-the-columns-of-the-cluster-view)?

In principle [cluster labels are shown in the cluster view](https://phy.readthedocs.io/en/latest/clustering/#using-cluster-labels) and support sorting and filtering. If that doesn't work please let me know.

Charles-Elliott, 2020-02-26:
Ah, thank you for the plugin, I hadn't noticed it somehow.

As for the label, apologies, I was referring to the manual cluster assignment (shown in green/grey/black).

rossant, 2020-02-26:
For the cluster groups, it's not yet possible to sort by them, but you can filter by typing `group=="good"` in the filter box at the top of the cluster view

Charles-Elliott, 2020-02-26:
So I figured. Thanks for your reply!