cortex-lab/phy #959: custom_split plugin: 'Supervisor' object has no attribute 'edit_actions'

ondracej, 2020-02-23:
Hello, I am beginning to work with the plugins for Phy 2, and I am running into some trouble with the "custom_split" plugin example. 

at line 17 of custom_split:  @controller.supervisor.edit_actions.add(shortcut='s')
I get the error:  'Supervisor' object has no attribute 'edit_actions' when I call the phy template-gui

Is there another way to add the keyboard shortcut to the gui? Or another workaround?

Also, is this k-means splitting actually already implemented in the "split_init" shortcut (Cntrl+Shift+K) found under the 'Edit' menu? What is split_init actually doing?

Thanks a lot for your help with all of this!

rossant, 2020-02-26:
Thanks for the bug report, this should be fixed now. In your plugin you should replace "edit_actions" by "actions". Let me know if that fixed it.

Also, split_init splits according to the original template assignments and does not perform any clustering algorithm.

ondracej, 2020-02-26:
Dear Cyrille, 

Thanks so much, this fixed it!  I also needed to install 'sklearn' in the env as documented in this issue: https://github.com/cortex-lab/phy/issues/927 - now it all works like a charm! 

And thanks for the explanation for the split_init command! 

All the best, Janie

closed 2020-02-26 by ondracej