cortex-lab/phy #960: the gui doesn't launch

Pascale42, 2020-02-25:
Hello, 

I have some trouble with opening the params.py files: the template gui does not launch (for old and new files)
I reinstalled python 3.7 using Anaconda 2019.10 for my windows 64-bit system, and installed a fresh phy2 env.
Unfortunately, the command "phy template-gui params.py --debug" doesn't display a thing.
Any idea on what is wrong? 

Best !

rossant, 2020-02-26:
Any error message in the terminal?

Pascale42, 2020-03-04:
nope, just... nothing

rossant, 2020-03-04:
could you post a screenshot of your screen after you launched "phy template-gui params.py --debug" ?

Pascale42, 2020-03-04:
Here it is: 
[image]

Pascale42, 2020-03-04:
I've tried to run it with a couple of different files that were generated (and analyzed) a year ago
This is the same outcome for each of them, even for recent files where Kilosort was run few days ago

rossant, 2020-03-04:
OK, can you try [this](https://github.com/cortex-lab/phy#running-phy-from-a-python-script)? By running the GUI from a Python script instead of the terminal you might see error messages

phanhuynh, 2022-02-22:
Hello everyone, 
I'm having the same issue and I tried the instructions on the link.
But the issue persists -- I believe it fails at this import:

`from phy.apps.template import template_gui`

Looking at the output of `print(dir(phy))` apps doesn't seem to be accessible :

`['Bunch', 'IPlugin', 'StringIO', '__author__', '__builtins__', '__cached__', '__doc__', '__email__', '__file__', '__loader__', '__name__', '__package__', '__path__', '__spec__', '__version__', '__version_git__', '_git_version', 'atexit', 'connect', 'discover_plugins', 'emit', 'get_plugin', 'load_master_config', 'logger', 'logging', 'on_exit', 'op', 'sys', 'test', 'unconnect', 'utils']`

Thank you so much in advance!