cortex-lab/phy #961: Can't Open Similarity View Window

SaberaTalukder, 2020-02-26:
We accidentally closed similarity view in phy and cannot reopen it (under the view tab there is no option to add it, and right clicking on any of the views in the window and trying to check the similarity view box doesn't work because it is greyed out ). We also recreated the phy2 conda environment and re-downloaded all of the packages which did not fix the problem. Interestingly the hotkeys for similarity view still work (pressing the space bar when a cluster is selected in cluster view will pull up a similar cluster and you can cycle through the data that way). Any help to get similarity view back would be much appreciated! Thanks!

rossant, 2020-02-26:
Go to `~/.phy/TemplateGUI/state.json` and make sure the field `SimilarityView` has a value of 1 in `view_count`

Do you use the latest version (2.0 beta)? If the similarity view is closable then it's a bug

SaberaTalukder, 2020-02-26:
We did that as well. When similarity view was accidentally closed the value was still 1 in the state.json. We then took the state.json from another computer with the phy setup and replaced the one on the machine where similarity view doesn't appear and it didn't fix the problem.

If we transfer the data to another machine that has the phy setup it opens with the similarity view.

rossant, 2020-02-26:
What if you rename state.json to state.json~?

SaberaTalukder, 2020-02-26:
That fixed it- thanks!

It opens with all possible views. Then if we delete all of the views that we don't want it reopens in the way we closed it last.

What exactly does adding the ~ to the end of the state.json do?

rossant, 2020-02-27:
It's like deleting it, but keeping a backup just in case!

closed 2020-02-27 by SaberaTalukder