cortex-lab/phy #965: GUI geometry not properly restored

szapp, 2020-02-28:
The positions of the views is not properly restored for views detached out of the main window. This seems to happen both on Windows and Linux.

When faulty plug-ins cause an error box on startup, the GUI geometry is recovered correctly. So I assume there might be something wrong with the order of opening the GUI and restoring its geometry. The error box seems to trigger restoring the geometry before showing the GUI.

Since the message box "solves" this problem, the changes below are my temporary workaround (an auto-closing message box before showing the GUI). This is of course not a good solution, but might help in pinpointing the underlying issue:

```diff
--- a/phy/apps/template/gui.py
+++ b/phy/apps/template/gui.py
@@ -197,6 +197,14 @@ def template_gui(params_path, **kwargs):  # pragma: no cover
     create_app()
     controller = TemplateController(model=load_model(params_path), dir_path=dir_path, **kwargs)
     gui = controller.create_gui()
+
+    # Work around to properly restore window geometry
+    from PyQt5.QtWidgets import QMessageBox
+    box = QMessageBox()
+    box.setStandardButtons(QMessageBox.Ok)
+    box.button(QMessageBox.Ok).animateClick(1)
+    box.exec_()
+
     gui.show()
     run_app()
     gui.close()

```
[Source](https://github.com/cortex-lab/phy/compare/master...szapp:keepgeometry#files)

rossant, 2020-03-02:
Okay, what about this, does it change anything?

```diff
diff --git a/phy/gui/gui.py b/phy/gui/gui.py
index fa640017..947aac5e 100644
--- a/phy/gui/gui.py
+++ b/phy/gui/gui.py
@@ -605,8 +605,8 @@ class GUI(QMainWindow):
 
     def show(self):
         """Show the window."""
-        emit('show', self)
         super(GUI, self).show()
+        emit('show', self)
 ```

szapp, 2020-03-02:
Yes, that works. Although only partially:

The views detached from the main GUI are restored at correct positions and sizes. The views within the GUI remain in their original arrangement, but their sizes do not seem to be maintained (this is not the case with the small "hack" I had used).