cortex-lab/phy #967: Problems in display - waveform view and feature view

Oryud, 2020-03-04:
Hello all,
I worked on my recordings on phy1, then moved to another computer with phy2.
Unfortunately, something is wrong with the waveform view and probably with the feature view as well, and they look completely different from how they looked in phy1, as if it is not alligned correctly so everything looks like noise - see attached images.
I guess this is a display problem. Does anyone know how to solve this?

Thanks a lot!
Or
[image]
[image]
[image]
[image]

rossant, 2020-03-04:
I think you need to update phy and phylib to the latest versions on GitHub

Oryud, 2020-03-05:
This seems to do the trick. Thank you so much!

closed 2020-03-05 by Oryud