cortex-lab/phy #971: "Illegal instruction 4" error on mac El Capitan with the latest installation.

tachyian, 2020-03-09:
I recently updated my phy2 installation on my Mac following the same steps as a new installation by installing Anaconda 3.7 (latest version, 2019.10) and then by invoking the following commands:
"conda create -n phy2 python pip numpy matplotlib scipy h5py pyqt cython pillow -y
conda activate phy2
pip install phy --pre --upgrade"
as described in the installation instruction. Now, when I go to the data folder and try to run phy as usual, I always get "Illegal instruction 4" error without any clue and can't go any further. "--debug" option doesn't show anything and I have no clue at all. phy2 used to work flawlessly before I tried the new installation.
My mac has El Capitan 10.11.6.
Any help will be really appreciated!

sshanbhag, 2020-03-25:
I'm having the same issue on my Mac, running El Capitan 10.11.6. 

(phy2) pangolin:data3 sshanbhag$ phy template-gui data3.npy 
Illegal instruction: 4

I will try installing an older version and see if that works.

rossant, 2020-03-25:
`(phy2) pangolin:data3 sshanbhag$ phy template-gui data3.npy`

the argument to phy template-gui should be a `params.py` file instead, right? Do you have the same error then?

tachyian, 2020-04-29:
Is there any update?
I've also tried installing a few different versions of Anaconda from the latest to older ones including Anaconda3-5.3.1-MacOSX-x86_64.sh, which used to work with phy2 last year. None of them works with phy2 any more immediately spitting out the same error: "Illegal instruction 4".
It seems that pretty much all other commands and tools that came with the Anaconda installation work without an error. So, I suspect it might be something that's related to phy2 and older version of Mac OSX, but I'm not an expert....
Can you please take a look and give any hint to troubleshoot it?
Thanks!