cortex-lab/phy #972: some kwik-gui problems with visualisation and shortcuts

reinderdorman, 2020-03-10:
Loading .kwik file, using phy2, upgraded, installed klusta in phy2 env as well. removed the .phy folders in users folder aswell as data folder. Windows 10, fresh anaconda install, phy2 installed 10-3-2020, kwik-gui seems to run on the GPU

Several things seem to go wrong, can't really put a finger on what the core problem is. Starting fresh, no .phy folders, loading a newly clustered .kwik file. kwik-GUI opens several windows, looking okay.

first problems: selecting a cluster in similarityView is not showed in other windows. Closing the waveformView window and re-opening via View-Waveform seems to work at one occassion However, at another point a moment ago, this opened a second waveformView, called WaveformView (1), which did not respond to keyboard shortcuts. Closing this window and reopening on one occassion resulted in WaveformView(2), and next (3), (4), etc. i.e. seems like the original WaveformView window is still active somewhere?

Furthermore, at several moments, when original WaveformView window was activa (i.e. without a number), the window didn't respond to keyboard shortcuts, but gives the following error in the terminal:
QAction::eventFilder: Ambiguous shortcut overload: Ctrl+Up
(or any other combination of ctrl/shirt + arrow keys).

As of just now, I once again removed the .phy folders. With freshly opening the GUI, all windows did not show any information. Closing and re-opening seems to work this time.

My apologies if this seems a bit confusing and non-reproducible, as I can't event confidently reproduce some of this behavior myself. But I thought i'd just post anyway to let you guys know that somewhere, something seems to be a bit off. Would love to think allong in trying to get any potential bugs sorted out.

Extra note: I seemed to work okay for now, However after a minute I noticed the WaveformView was not updating anymore, when selecting different clusters. Closing, reopening said window did results in the window displaying the appropriate clusters (both best and similar onces), but now the short-cut error is once again back. note, the shortcut's do not work, but shrinking etc via the little menu does still work)

rossant, 2020-03-10:
Try the `--clear-state` option at the end of the `phy template-gui...` command line to refresh the view layout.
Any error message in the terminal?

reinderdorman, 2020-03-10:
Thanks for the tip. Before I forgot, I'm using kwik-gui, not template-gui (but I think/hope that for gui visualization problems this probably won't matter much?)

No error messages.

As expected, the gui is reset to default layout, and now all the windows except the traceview are empty. selecting any cluster doesn't do anything.

However, if I now close the waveFormView, and open a new window via view > add waveformview, this new waveform window is working again, showing best, similar clusters properly. 

After removing and then adding the FeatureView window, this new FeatureView window also properly displays the features for the selected clusters. However, now the WaveformView window is not updating anymore when I select another clusters. 

Also, using hotkeys results in the same QAction::eventFilter message as poster above. 

I closed and tried the same command again, and now it starts up with all the windows displaying something. Selecting clusters and hotkeys seem to work normally. However, if I close a window (in this case ProbeView), it stops updating all windows when I select different clusters. Hotkeys work normally. This seems to be pretty reliable behavior, it happends every time I launch the kwik-GUI, and when when I close ClusterScatterView, TraceView too, probably every window

[image]

rossant, 2020-03-10:
What if, instead of closing the views, you undock them from the main window, to make them floating windows, and redock them?
What is your graphics card?
There were somewhat related problems with Intel drivers on Windows, not sure if that's related.

reinderdorman, 2020-03-10:
Undocking seems to work fine. Tried with the WaveformViewer. Out-docked, it displays the correct clusters, hotkeys work fine. Re-docking, everything works fine too. 

Yes I've read about the Intel problems. I'm using a nvidia GeForce GTX 1050, and i'm running python on the GPU (according to some nvidia control program). I would think that this means it's using the GPU for the kwik-gui aswell