cortex-lab/phy #978: ImportError

AhnafHassan, 2020-03-14:
Hi, 

I have been using Kilosort and Phy2 for a few months. But today when I was trying to open Phy2, I kept getting the following error messages. 

[image]

Kindly help me out. Thanks in advance. 

Sincerely,
Ahnaf

rossant, 2020-03-14:
Hi, it seems you're using the dev branch of phy. I'd suggest to switch to the master branch which is now the stable one.
The alternative is to install the dev branch of phylib as it is required by the dev branch of phy.

AhnafHassan, 2020-03-14:
Hi Rossant, 

Thank you for your response. So, to solve this, I basically need to type this: 
pip install git+https://github.com/cortex-lab/phy.git@master

instead of pip install git+https://github.com/cortex-lab/phy.git@dev

Is that correct? Thank you in advance. 

Sincerely, 
Ahnaf

rossant, 2020-03-14:
Yes. You might need to add `--upgrade` at the end of the command as well. Let me know if that worked.

AhnafHassan, 2020-03-16:
Yep, that solves it. Thanks.

closed 2020-03-16 by AhnafHassan