cortex-lab/phy #979: GUI not displaying data

mgcampb, 2020-03-18:
Hi Cyrille,

I just downloaded Phy2 onto a new machine. I use it regularly on other machines. When I open the GUI, the tables are fine, all of the data windows are blank (see attached screenshot). The machine has an NVIDIA GeForce 1080 GPU. I tried updating the driver and it didn't help. Any ideas?

Thanks!
Malcolm
[image]

mgcampb, 2020-03-18:
Just saw this issue in Kilosort2's github:

https://github.com/MouseLand/Kilosort2/issues/195

My issue appears to be identical. Same errors with the --debug option.

takacsflora, 2020-03-23:
Hi Cyrille, 

I have the same issue with an NVIDIA GeForce GTX 1060 6GB GPU. I have tried uninstalling and reinstalling phy, updating driver via device manager, and also updating driver in GeForce Experience with the automatic update (which installed GeForce Game Ready Driver 442.74), but no success. 

Thanks for your help,
Flora

rossant, 2020-03-23:
Do you have a way to double check that phy is using the NVIDIA GPU and not an integrated graphics driver that may coexist with the GPU?

mgcampb, 2020-03-23:
In my case the NVIDIA GPU is the only graphics card installed in the machine.

takacsflora, 2020-03-23:
Same. Also it seems it is correctly accessed from my phy environment. I forgot to add that for me this is not a new installation issue - I had phy running on my computer before, and it stopped working only recently (I only had iblapps and ibllib as new packages added to my computer since then). 
[image]

jaib1, 2020-03-23:
For Flora's case could this actually be an issue with pyqt5? Is it possible that the pyqt5 dependency required by the iblapps repo is somehow interacting with and causing issues with phy?

Idavr, 2020-08-03:
I seem to have the exact same problem. It was working perfectly when I first installed it, but now suddenly (when revisiting some analyzed data sets by Kilosort2) the windows are completely empty. Anyone found a solution to this?

neurongeek, 2020-08-04:
Two things solved it for me -
1. Update your NVDIA and intel drivers to the latest version
2. install vispy in your phy environment using pip (pip install vispy)

hope it works!

-Suyash

Idavr, 2020-08-04:
I seriously shouted out of joy right now, this got it to work! Thank you so much @neurongeek ! 👍