cortex-lab/phy #981: Plotting the templates used for sorting

AhnafHassan, 2020-03-21:
Hi, 

I am trying to plot the mean spike waveform as well as the template used to detect a cell. I have a 128-channel probe, and my number of desired clusters was 256. Therefore, templates.npy (256 by 82 by 128) contains 256 templates for each channel, since templates are defined on all channels. My question is: for a particular cell, how do I know which of these 256 templates was used so I can generate the aforementioned plot? 

Thanks in advance. 

Sincerely, 
Ahnaf

P.S. There was a similar issue before (https://github.com/cortex-lab/KiloSort/issues/35), but the answer probably was not elaborate to me.

rossant, 2020-03-23:
Hi, a work in progress documentation page is available here: https://github.com/cortex-lab/phy/blob/dev/docs/analysis.md
it only works on the dev branches of phy and phylib at the moment

AhnafHassan, 2020-04-23:
Thank you. 

Is there any way to plot the principal components in MATLAB similar to Phy? (I read pc_features.npy in MATLAB as per: https://github.com/kwikteam/phy-contrib/blob/master/docs/template-gui.md, but cannot figure out what the 3rd index will be)