cortex-lab/phy #982: Problem in waveformview phy2

fveen, 2020-03-23:
Can't get the waveform view to work normally, increasing and decreasing is impossible, so I can't visualize the waveform.  But in the meantime I'm not getting an error. All the other views function well. Does anyone know how to fix this? reset this?

rossant, 2020-03-23:
Could you post a screenshot?

fveen, 2020-03-24:
In the meantime uninstalled the new version and installed the old phy, same problem.. see issue 983

brmullen, 2021-12-09:
I am having this same issue
[image]
[image]
.