cortex-lab/phy #983: Problem with visualizing the waveform in waveformview Phy

fveen, 2020-03-24:
Can't get the waveform view to work normally I can't visualize the waveform. But in the meantime I'm not getting an error.  Had the same problem with phy2 so now installed the older phy instead.  But get the same problem here. All the other views function well. Does anyone know how to fix this? reset this?

rossant, 2020-03-24:
Could you post a screenshot of the entire GUI?

fveen, 2020-03-24:
[image]

rossant, 2020-03-24:
Thanks. Could you delete ~/.phy/KwikGUI/state.json and try again?

fveen, 2020-03-24:
[image]

rossant, 2020-03-24:
do this instead:

`del c:\users\felic\.phy\KwikGUI/state.json`

fveen, 2020-03-24:
Thank you for your help! Then I get this error:

(kk_ff) C:\Users\felic\OneDrive\Documenten\RD\Klusta_Probe1_15012020>del c:\users\felic\.phy\KwikGUI/state.json
Parameter format not correct - "state.json"

rossant, 2020-03-24:
sorry I did a mistake, do this instead:

`del c:\users\felic\.phy\KwikGUI\state.json`

fveen, 2020-03-24:
Yes worked and Gui does look different, but after some trying still can't get the waveform right:

fveen, 2020-03-24:
[image]

fveen, 2020-03-24:
[image]

rossant, 2020-03-24:
Okay  there may be something wrong in the file. Do you have a way to send me your dataset? (kwik and kwx files) via dropbox or similar?

fveen, 2020-03-25:
I wish it was that..but the same file works fine on another computer.

closed 2020-03-25 by fveen