cortex-lab/phy #984: Phy2 GUI freezes cluster display

mari-sosa, 2020-03-24:
I'm using phy2 on Ubuntu 18.04 (installed using pip) and am having a graphics issue where partway through the session, the views will get "stuck" on a single cluster and fail to update when I select new clusters.  

This is the cluster I was viewing before the freeze (cluster 2):
[image]

Note no change in the display when I select another cluster (cluster 156) or its nearest neighbor in Similarity View:
[image]

Note that there are no errors reported during this freeze using the `--debug` option:
[image]

The gui is still operational, just doesn't update.  I'm using an NVIDIA GeForce RTX 2070 Super and nvidia driver 440.64. Any ideas?

mari-sosa, 2020-03-25:
For now this seems to be resolved by deleting the file ~/.phy/TemplateGUI/state.json and re-launching phy.

rossant, 2020-04-06:
okay feel free to reopen if the problems appears again

closed 2020-04-06 by rossant